introlab/rtabmap #1380: Segmentation fault saving to .obj

MatthijsRN, 2024-11-13:
Hi folks,
I run rtabmap in vscode with extension dev container and use the noble container. I make the tools/export/main.cpp without errors. However when I run it it gives a segmentation fault in line 1979 (success = pcl::io::saveOBJFile(outputPath, *textureMesh) == 0;)
The issue persists when using different .db files. It can successfully export .ply files. 

Do you have any suggestions? 

I run with the arguments:
"args": ["--texture", "--texture_size", "4096", "--texture_range", "3", "--poisson_depth", "9", "--max_polygons", "200000", "--color_radius", "0",  "--no_clean",  "~/Documents/RTAB-Map/cps18_data/office/20241111/rtabmap/241111-115011.db"]
Attached is my devcontainer.json. I run on a Ubuntu 22.04 machine.

[devcontainer.json](https://github.com/user-attachments/files/17730802/devcontainer.json)

MatthijsRN, 2024-11-14:
The issue persists on Ubuntu 24 machine. The issue is not there when running the jammy dockerfile

matlabbe, 2024-11-16:
Since PCL 1.13, the order of the polygons changed. It was fixed in the UI->Export Clouds, but not for rtabmap-export. 
https://github.com/introlab/rtabmap/blob/b2e26e81c14bd7a1394da10998d0264968089cbe/guilib/src/ExportCloudsDialog.cpp#L5118-L5132

The commit above should fix it.

commit b2e26e81 (2024-11-16): Fixed #1380
closed 2024-11-16 by matlabbe