introlab/rtabmap #1382: Visualizer: pointcloud map, not an octomap.

taehun-ryu, 2024-11-15:
I'm grateful for your wonderful work. I have one question about rtabmap's origin Visualizer, not rviz in rtabmap_ros.

**The '3D map' window shows global octomap and local point clouds. However, in my case, I don't need to visualize octomap. I only wish to have a 3D point cloud map in the '3D Map' window.**

Is there any way to visualize a global 3D point cloud map?

Thanks.

closed 2024-11-15 by taehun-ryu

matlabbe, 2024-11-17:
With default rtabmap_viz's settings, the OctoMap is not shown by default, only the point cloud using Preferences->3D rendering settings.

If for some reasons the OctoMap is shown, you can disable it under Preferences-> 3D Rendering->GridMap Assembling panel (uncheck Octomap checkbox).

taehun-ryu, 2024-11-18:
Sorry for the confusion by opening and closing the issue window multiple times. I didn't know whether to ask this question in the issue or the rtabmap forum. **Thank you very much for your answers in both places.**

I have one more question. I ran ros2 launch rtabmap_examples euroc_datasets.launch.py. In that case, it seems that octomap is visualized. Can I turn off octomap not through rtabmap_viz's GUI menu window (e.g. Preferences-> 3D Rendering->GridMap), but through rtabmap parameter or code modification?