introlab/rtabmap #1385: Segmentation Fault with 3D map view

jared-cormier, 2024-11-18:
Any interaction or change in the 3D Map window causes a segmentation fault. I suspect this has something to do with VTK9.1, but I haven't worked with C++ in about 10 years, so I'm a little rusty.  Logging was pretty useless unless there's a logging flag not in the main preferences screen. Haven't had time to build all the dependencies from source to really nail down where the issue is occurring, but figured I'd log this until I can get more time to dive into it.

Configuration:
- MOBILE_BUILD=OFF
-- Found OpenMP: TRUE (found version "4.5") found components: CXX 
-- Found flann version 1.9.2
-- Found Qhull version 8.0.2
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found SQLite3: /usr/include /usr/lib/aarch64-linux-gnu/libsqlite3.so
-- Found OpenMP: TRUE (found version "4.5") found components: C CXX 
-- Found OpenMP: /usr/lib/gcc/aarch64-linux-gnu/11/libgomp.so;/usr/lib/aarch64-linux-gnu/libpthread.a
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/local/include/pcl-1.14;/usr/include/eigen3;/usr/include
-- Found ZLIB: /usr/include
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- VTK>=9 detected, will use VTK_QT_VERSION=5 for Qt version.
-- VTK_RENDERING_BACKEND=
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o (targets)
-- g2o include dir: /usr/local/include/g2o/core;/usr/local/include
-- Latest g2o factory version detected with shared ptr (factory file: /usr/local/include/g2o/core/factory.h).
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Architecture: aarch64
-- Found Pthreads
-- yaml-cpp not found, euroc_dataset tool won't be built...
CMake Deprecation Warning at examples/LidarMapping/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.8
--   CMAKE_VERSION =        3.25.1
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0 -fopenmp -std=c++17
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_HAS_CXX17_OVERALIGN=0
--   PCL_VERSION = 1.14.1
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.6.0 xfeatures2d = NO, nonfree = NO (License: Apache 2)
--   With Qt 5.15.8            = YES (License: Open Source or Commercial)
--   With VTK 9.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (WITH_TORCH=OFF)
--   With Python3              = NO (WITH_PYTHON=OFF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
--   With libLAS               = NO (WITH_LIBLAS=OFF)
--   With CudaSift             = NO (WITH_CUDASIFT=OFF)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o 1.0.0            = YES (License: BSD)
--  *With GTSAM                = NO (GTSAM not found)
--  *With Ceres                = NO (WITH_CERES=OFF)
--   With MRPT                 = NO (MRPT not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (WITH_CVSBA=OFF)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With CCCoreLib            = NO (WITH_CCCORELIB=OFF)
--   With Open3D               = NO (WITH_OPEN3D=OFF)
--   With OpenGV               = NO (OpenGV not found)
-- 
--  Reconstruction Approaches:
--   With OctoMap              = NO (octomap not found)
--   With GridMap              = NO (grid_map_core not found)
--   With CPUTSDF              = NO (WITH_CPUTSDF=OFF)
--   With OpenChisel           = NO (WITH_OPENCHISEL=OFF)
--   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
-- 
--  Camera Drivers:
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI               = NO (OpenNI not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
--   With XVisio SDK           = NO (WITH_XVSDK=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done

jared-cormier, 2024-11-19:
Update: added OctoMap as that looks like a conditional dependency, no change; next steps: build older VTK versions and recompile.

jared-cormier, 2024-11-20:
https://github.com/introlab/rtabmap/blob/b2e26e81c14bd7a1394da10998d0264968089cbe/guilib/src/CloudViewer.cpp#L3808
One of the failing lines. I did discover, as part of my spamming of log messages, I did find there were other locations which I shall try to identify the exact locations of prior to attempting fixes.

matlabbe, 2024-11-24:
You are using PCL 1.14.1 built from source. That could be related to that. Are you using Ubuntu 22.04 or 24.04?

jared-cormier, 2024-11-24:
> You are using PCL 1.14.1 built from source. That could be related to that. Are you using Ubuntu 22.04 or 24.04?

24.04, but I've also tried to get this running on a Raspberry Pi 5 with oddly similar results.

matlabbe, 2024-11-24:
I am currently investigating what is happening with building latest PCL with c++17. Could also be related to this: https://github.com/PointCloudLibrary/pcl/issues/6126

matlabbe, 2024-11-24:
So I updated this [Dockerfile](https://github.com/introlab/rtabmap/blob/master/docker/latest_deps/Dockerfile) to show a working example when building latest PCL version on recent machines. In particular, to be most portable, try building PCL with:
```
cd pcl/build
cmake -DCMAKE_CXX_STANDARD=17 \
   -DBUILD_tools=OFF \
   -DPCL_ENABLE_AVX=OFF \
   -DPCL_ENABLE_MARCHNATIVE=OFF \
   -DPCL_ENABLE_SSE=OFF ..
```

jared-cormier, 2024-11-28:
I'm encountering the exact same issue with a clean image Raspberry Pi 5 running Raspbian and only official binaries for dependencies. (Wayland doesn't work, I had to switch the UI to X prior to it even loading)
CMAKE output below: 
`-- MOBILE_BUILD=OFF
-- Eigen found (include: /usr/include/eigen3, version: 3.4.0)
-- OpenNI found (version: 1.5.4.0, include: /usr/include/ni, lib: /usr/lib/libOpenNI.so;libusb::libusb)
-- Eigen found (include: /usr/include/eigen3, version: 3.4.0)
-- Found flann version 1.9.2
-- Found Qhull version 8.0.2
-- OpenNI found (version: 1.5.4.0, include: /usr/include/ni, lib: /usr/lib/libOpenNI.so;libusb::libusb)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found SQLite3: /usr/include /usr/lib/aarch64-linux-gnu/libsqlite3.so
-- Found OpenMP: /usr/lib/gcc/aarch64-linux-gnu/12/libgomp.so;/usr/lib/aarch64-linux-gnu/libpthread.a
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.13;/usr/include/eigen3;/usr/include;/usr/include/ni
-- Found ZLIB: /usr/include
-- VTK>=9 detected, will use VTK_QT_VERSION=5 for Qt version.
-- VTK_RENDERING_BACKEND=
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Could NOT find g2o (missing: g2o_DIR)
-- RTAB-Map's cmake g2o find module used for convenience (for older g2o versions)
CMake Warning (dev) at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (CSPARSE)
  does not match the name of the calling package (G2O).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake_modules/FindG2O.cmake:9 (find_package_handle_standard_args)
  CMakeLists.txt:516 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find CSPARSE (missing: CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Architecture: aarch64
-- Found Pthreads
-- yaml-cpp not found, euroc_dataset tool won't be built...
CMake Deprecation Warning at examples/LidarMapping/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.8
--   CMAKE_VERSION =        3.25.1
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0 -fopenmp -std=c++14
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG
--   PCL_VERSION = 1.13.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.6.0 xfeatures2d = NO, nonfree = NO (License: Apache 2)
--   With Qt 5.15.8            = YES (License: Open Source or Commercial)
--   With VTK 9.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
[... 38 lines cut ...]
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
--   With XVisio SDK           = NO (WITH_XVSDK=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jeep/rtabmap/build
`

matlabbe, 2024-12-02:
Could it be similar to this issue? https://github.com/introlab/rtabmap_ros/issues/1230 (or this comment https://github.com/introlab/rtabmap/issues/1062#issuecomment-1597997914)

I would build without Qt on RPI (`-DWITH_QT=OFF`) if you are planning to use ROS.

jared-cormier, 2024-12-02:
> Could it be similar to this issue? [introlab/rtabmap_ros#1230](https://github.com/introlab/rtabmap_ros/issues/1230) (or this comment [#1062 (comment)](https://github.com/introlab/rtabmap/issues/1062#issuecomment-1597997914))
> 
> I would build without Qt on RPI (`-DWITH_QT=OFF`) if you are planning to use ROS.

Quite probably related; I won't be using ROS as I am only interested in utilizing the point clouds for now. I see lots of compiling from source in my future. FWIW, the RPi 4 and RPi 5 CPU and GPU performance difference is pretty stark. But yes, I won't be using it as my final platform (AGX Orin + cameras + LiDAR).

Joseph-Momma, 2024-12-19:
I have pretty much the exact same problem with raspberry pi 5. I have used the ROS version, compiled rtabmap standalone from source, installed on raspberry-pi OS and there is no difference. It works just fine on my x86 laptop. I have noticed that the 3D Map can populate with points, but the moment I move or scroll around in the map, I get a segmentation fault. At first I was concerned that it was the raspberry pi that was too slow to handle rendering the 3D map in real time, but I have since found numerous videos of people using rtabmap on pi4,  rendering the 3D map in real time. 

I am on Ubuntu 24.04

[image]

```
-- Info :
--   RTAB-Map Version =     0.21.6
--   CMAKE_VERSION =        3.28.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0 -fopenmp -std=c++14
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_HAS_CXX17_OVERALIGN=0
--   PCL_VERSION = 1.14.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.6.0 xfeatures2d = NO, nonfree = NO (License: Apache 2)
--   With Qt 5.15.13            = YES (License: Open Source or Commercial)
--   With VTK 9.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (WITH_TORCH=OFF)
--   With Python3              = NO (WITH_PYTHON=OFF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
--   With CudaSift             = NO (WITH_CUDASIFT=OFF)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o                  = NO (g2o not found)
--  *With GTSAM                = NO (GTSAM not found)
--  *With Ceres                = NO (WITH_CERES=OFF)
--   With MRPT                 = NO (MRPT not found)
--   With VERTIGO              = NO (GTSAM or g2o required)
--   With cvsba                = NO (WITH_CVSBA=OFF)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With CCCoreLib            = NO (WITH_CCCORELIB=OFF)
--   With Open3D               = NO (WITH_OPEN3D=OFF)
--   With OpenGV               = NO (OpenGV not found)
-- 
--  Reconstruction Approaches:
--   With OctoMap              = NO (octomap not found)
--   With GridMap              = NO (grid_map_core not found)
--   With CPUTSDF              = NO (WITH_CPUTSDF=OFF)
--   With OpenChisel           = NO (WITH_OPENCHISEL=OFF)
--   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
-- 
--  Camera Drivers:
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI               = NO (OpenNI not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI 2.28.0       = YES (License: MIT)
--   With XVisio SDK           = NO (WITH_XVSDK=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
```