introlab/rtabmap #1387: Difference in poses export between GUI and `rtabmap-export` script.

hkortier, 2024-11-23:
Hi,
I consider a difference in output when I want to export the poses. I use the following options for the GUI (database viewer):

RGBD-SLAM ID format -> Map's graph -> robot (base frame)

And for the script I use `rtabmap-export --poses --poses_format 11 $DB`

It seems that the script version does some optimization:

```
[ INFO] (2024-11-23 10:12:38.628) Rtabmap.cpp:5131::optimizeCurrentMap() Optimize map: around location 1611 (lookInDatabase=true)
[ INFO] (2024-11-23 10:12:38.629) Rtabmap.cpp:5140::optimizeCurrentMap() get 1579 ids time 0.000964 s
[ INFO] (2024-11-23 10:12:38.632) Rtabmap.cpp:5183::optimizeGraph() get constraints (ids=1579, 1583 poses, 3253 edges) time 0.002875 s
[ INFO] (2024-11-23 10:12:38.701) Rtabmap.cpp:5255::optimizeGraph() Optimization time 0.068538 s
[ INFO] (2024-11-23 10:12:38.701) Rtabmap.cpp:5143::optimizeCurrentMap() optimize time 0.072313 s
[ INFO] (2024-11-23 10:12:38.702) Rtabmap.cpp:5152::optimizeCurrentMap() Correction (from node 1611) xyz=-0.011847,-0.096141,-0.002846 rpy=0.000646,-0.000955,0.029272
Optimizing the map... done (0.782533s, poses=1583).
Create and assemble the clouds...
Create and assemble the clouds... done (4.187351s, 3020073 points).
Poses exported to
```

The outputs are different (global rotation) and tiny pose differences. In addition the GUI version export the landmark poses as well. I use the latest osx app (GUI) and latest source code (script).

Can you explain the difference? Is it deliberately?

matlabbe, 2024-11-24:
rtabmap-export and rtabmap-databaseViewer export poses differently. rtabmap-export re-optimizes the graph before exporting them. rtabmap-databaseViewer exports the poses that are currently seen in View->GraphViewer, with selected iteration and type:

[image]

Support for landmarks have been added in this commit https://github.com/introlab/rtabmap/commit/13cd5e7a5e4ccf248a8397548bb2499fe6238e01 for the viewer, but not export cli. I am working on a branch refactoring export tool, I'll add it as todo in it. https://github.com/introlab/rtabmap/pull/1391 In that branch, I added `--opt` option to export optimized poses of the map, which would be exactly the same than if you select Local Optimized graph in the viewer. Otherwise, when the graph is re-optimized, there could be small variations.

hkortier, 2024-11-25:
Thanks for the clarification. Including a landmark export in the cli tool would be a nice add-on!

Is the re-optimization step before export using the cli tool necessary or could that be made optional?

matlabbe, 2024-12-02:
> Is the re-optimization step before export using the cli tool necessary or could that be made optional?

If `--opt` is set, it won't be re-optimized, unless there are now optimized poses saved in the database.

closed 2025-03-15 by hkortier