introlab/rtabmap #1389: Error While Integrating

FrankFourth, 2024-11-23:
I am trying to integrate library into my project, but I am encountering some errors during the process.Additionally, I might have made mistakes during the process. Could you please guide me on the correct steps I should follow for the integration? Your assistance would be greatly appreciated.

[image]

FrankFourth, 2024-11-23:
There were no errors during the execution of the install.sh script, and I am using Xcode 14. I don’t encounter any issues when using the original downloaded project. However, when I try to set it up from scratch in a new project, I face these errors.

FrankFourth, 2024-11-23:
When I re-added the frameworks as content, the number of errors decreased to 72, and they changed as follows, but I still can’t figure out where I went wrong during the setup.

[image]

matlabbe, 2024-11-24:
In your iOS project config, you need to add all libraries created by install.sh script as dependencies. Open the original project here and compare the configs with yours.

ioridev, 2024-12-11:
[image]
You need to add libpng etc etc etc here