introlab/rtabmap #1392: Depth Image

dscho15, 2024-11-27:
[image]

Hi, this might be a very basic question, but I downloaded the RTAB-Map app and created a scan. I managed to extract a depth photo from the .db file, but I haven’t been able to decode the absolute depth from the depth map. Has anyone here successfully done this? If so, could you please share the mapping?

matlabbe, 2024-12-02:
If you extract the blob from the database, the format is PNG. You can use opencv [imdecode](https://www.geeksforgeeks.org/python-opencv-imdecode-function/) (you should use flag to not change format to color) function directly on that data to recover an opencv matrix. The matrix format would be U16C1, depth in mm. You can divide by 1000 to get the depth in meters.

dscho15, 2024-12-04:
First of all, thanks for taking your time. Much appreciated!

I've tried several different settings neither of them works  (I'll share my code down here). For me it seems like the photo is U8C4 or perhaps U16C2 and not U16C1. Any suggestions?

`from peewee import *

import datetime
import io
import numpy as np
import cv2

from PIL import Image
from pathlib import Path

db = SqliteDatabase('PATH_TO_DB_FILE')
db.connect()

class BaseModel(Model):
    class Meta:
        database = db
    
class Data(BaseModel):
    id = IntegerField(primary_key=True)
    image = BlobField()
    depth = BlobField()
    calibration = BlobField()
    scan = BlobField()
    scan_info = BlobField()
    ground_cells = BlobField()
    obstacle_cells = BlobField()
    empty_cells = BlobField()
    cell_size = FloatField()
    view_point_x = FloatField()
    view_point_y = FloatField()
    view_point_z = FloatField()
    user_data = BlobField()
    time_enter = DateTimeField(default=datetime.datetime.now)
    
def extract_rows_from_table(model):
    return list(model.select())

rows = extract_rows_from_table(Data)

image = io.BytesIO(rows[0].depth)
with open("test.png", "wb") as f:
    f.write(rows[0].depth)

image = np.asarray(bytearray(rows[0].depth), dtype="uint8")
image = cv2.imdecode(image, cv2.IMREAD_UNCHANGED)

image = np.asarray(bytearray(rows[0].depth), dtype="uint8")
image = cv2.imdecode(image, cv2.IMREAD_ANYDEPTH)
`

matlabbe, 2024-12-07:
Oh yeah, on iOS the depth image is 32FC1 type, which is encoded in 8UC4 format to be able to encode in PNG.
https://github.com/introlab/rtabmap/blob/d33e98a540fbea0d56dd9ddb8df7f9003229983f/app/android/jni/RTABMapApp.cpp#L3871-L3872

You have to recast the 8UC4 into a 32FC1 matrix. The depth would be in meters. Here what we do in c++:
https://github.com/introlab/rtabmap/blob/d33e98a540fbea0d56dd9ddb8df7f9003229983f/corelib/src/Compression.cpp#L140-L147

dscho15, 2024-12-09:
Thanks! 

Here's the code for decoding it using PIL and numpy tricks 

```
from peewee import *

import datetime
import io
import numpy as np
import cv2

from matplotlib import pyplot as plt

from PIL import Image
from pathlib import Path

# Initialize the database
db = SqliteDatabase('241102-23114 PM.db')
db.connect()

# Define the BaseModel
class BaseModel(Model):
    class Meta:
        database = db
        
class Feature(BaseModel):
    node_id = IntegerField()
    word_id = IntegerField()
    pos_x = IntegerField()
    pos_y = IntegerField()
    size = IntegerField()
    dir = IntegerField()
    response = FloatField()
    octave = IntegerField()
    depth_x = FloatField()
    depth_y = FloatField()
    depth_z = FloatField()
    descriptor_size = IntegerField()
    descriptor = BlobField()
    

# Define the Data model
class Data(BaseModel):
    id = IntegerField(primary_key=True)
    image = BlobField()
    depth = BlobField()
    calibration = BlobField()
    scan = BlobField()
    scan_info = BlobField()
    ground_cells = BlobField()
    obstacle_cells = BlobField()
    empty_cells = BlobField()
    cell_size = FloatField()
    view_point_x = FloatField()
    view_point_y = FloatField()
    view_point_z = FloatField()
    user_data = BlobField()
    time_enter = DateTimeField(default=datetime.datetime.now)

class Node(BaseModel):
    id = IntegerField(primary_key=True)
    map_id = IntegerField()
    weight = FloatField()
    stamp = DateTimeField(default=datetime.datetime.now)
    transform = BlobField()
    ground_truth_pose = BlobField()
    velocity = BlobField()
    label = TextField()
    gps = BlobField()
    env_sensors = BlobField()
    time_enter = DateTimeField(default=datetime.datetime.now)
    
# Function to extract rows as Data model instances
def extract_rows_from_table(model):
    return list(model.select())

# Fetch rows from the Data table
rows = extract_rows_from_table(Data)

depth = rows[0].depth
image = Image.open(io.BytesIO(depth))

depth_image = np.array(image)
h, w, c = depth_image.shape
depth_image = depth_image.reshape(h*w, c)
depth_image = depth_image[:, [2, 1, 0, 3]]
depth_image = depth_image.reshape(h, w, c)
depth_bytes = depth_image.tobytes()

image = np.frombuffer(depth_bytes, dtype=np.float32)

# save image
image = image.reshape(h, w)

plt.imshow(image)
plt.savefig("test.png")
```

closed 2024-12-09 by dscho15