introlab/rtabmap #1393: How is GraphView panel different than the published Map topics?

mvrius, 2024-11-30:
In RVIZ if I enable the View "Graph View", I get a Map equivalent which seems to me much better than what is being published on any of the *map topics.

How is the Graph View different, and can I essentially reproduce the same graph view as a map topic?

Thanks!

matlabbe, 2024-12-02:
I think you are referring to Graph View panel in rtabmap_viz, not RVIZ. The occupancy grid should be the same than the `/map`,`/grid_map` topic published by rtabmap node. Rviz should be able to show that map topic with same color. For the graph, we can show a similar graph with `rtabmap_rviz_plugins/MapGraph` plugin in rviz.

mvrius, 2024-12-02:
@matlabbe You are correct, I am referring to the Graph View panel in `rtabmap_viz`.

In my use case, the visualization presented there looks like a way more accurate map than what is in /map or /grid_map.

That's why I was asking what is being displayed there, because I would like to use that as a map, if possible, since it's seems to be accurate wrt how my space looks like, were as the other published maps don't have clearly defined edges. While the other topics\visualizations seem incomplete.

matlabbe, 2024-12-03:
Can you show a screenshot comparing each other? I am not sure to understand the differences as they should be identical.