introlab/rtabmap #1397: IOS: controlling on screen messages?

APOS80, 2024-12-04:
Hello there

it would be convenient to control what messages you see when scanning, it’s hard to know if a marker has been found or not.

matlabbe, 2024-12-07:
The order of messages are defined [here](https://github.com/introlab/rtabmap/blob/d33e98a540fbea0d56dd9ddb8df7f9003229983f/app/ios/RTABMapApp/ViewController.swift#L399-L427):

1. Visual Loop Closure Detected
2. Visual Loop Closure Rejected
3. Landmark Detected

The color of the background would also change depending of these [events](https://github.com/introlab/rtabmap/blob/d33e98a540fbea0d56dd9ddb8df7f9003229983f/app/android/jni/RTABMapApp.cpp#L1742-L1769):
1. Green = Visual Loop Closure Detected
2. Yellow = Proximity Loop Closure Detected
3. Orange = Landmark Detected
4. Dark Green = Visual Loop Closure Rejected
5. Blue = Rehearsal happened
6. Dark Magenta = Image filtered for too fast motion

So for a landmark, you should see a message telling which marker number has been detected, and the background should turn orange. In your case, is it because there are too many messages trying to show up at the same time (making difficult to read) or that landmark messages are missing?

APOS80, 2024-12-07:
I feel it’s to many messages in one spot and it’s sort of a war of the messages.

At least I would appreciate if I could set what messages are more important and how long they should stay at the screen. Or if they occurs at the same time, written beside each other.

And regarding the colours I mostly see red and orange. Maybe it would be easier to have dots showing in the corner of the screen, several dots at once if they happen at the same time.

Besides all this the app has really gotten better than last time I tried it. It’s nearly useful in my work as a construction surveyor.