introlab/rtabmap #1398: rtabmap-export  (Global Full vs Global iterative)

mattiasmar, 2024-12-05:
Hello,
I've noticed that the tool "rtabmap-export --poses" generates poses corresponding to what is referred to as "Global Full" in the rtabmap-databaseViewer.

Can one instead have rtabmap-export generate poses corresponding to  "Global iterative"? How?

matlabbe, 2024-12-07:
Related to https://github.com/introlab/rtabmap/issues/1387

Yes, it is Global Full. I'll check if I can also add the iterative option in that PR: https://github.com/introlab/rtabmap/pull/1391

mattiasmar, 2024-12-08:
During localization which graph is used: "Global iterative" or "Global full"?

matlabbe, 2024-12-10:
Full. Both are iterative. There should not be a huge difference, but with "iterative", we do each iteration outside the optimization library to get intermediate results (that is the default one in rtabmap-databaseViewer, to give a slider to see state of the graph after each iteration). The "full" approach lets the optimizer doing all the iterations internally. I don't have a full comparison between these two approaches, but rtabmap will use "full" when mapping/localizing.

closed 2024-12-10 by mattiasmar