introlab/rtabmap #1400: Failure to load calibration from ZED Open Capture

marius851000, 2024-12-07:
I recently bought a ZED mini camera that I plan to use with zed open capture. The program correctly fetch the calibration, but it appear that file parsing fail with the "ZED File invalid" error message.

The file, download from "https://www.stereolabs.com/developers/calib?SN=10025811", (renamed with a .txt to please GitHub) is:
[SN10025811.conf.txt](https://github.com/user-attachments/files/18048573/SN10025811.conf.txt)

I’ll try to debug this problem myself.

marius851000, 2024-12-07:
After some testing, it appears to be a locale issue, where atof use "," instead of "." for some reason.

matlabbe, 2024-12-08:
Is it that line causing the issue?
https://github.com/introlab/rtabmap/blob/d33e98a540fbea0d56dd9ddb8df7f9003229983f/corelib/src/camera/CameraStereoZedOC.cpp#L95

That could be fixed with `uStr2Float` function instead: https://github.com/introlab/rtabmap/commit/3d1eccbce1145596fea00c5b387538a0c83a5966

marius851000, 2024-12-09:
> That could be fixed with *uStr2Float* function instead: 3d1eccb[https://github.com/introlab/rtabmap/commit/3d1eccbce1145596fea00c5b387538a0c83a5966]

Yes. I've prepared a patch that use getDatas (from C++17), but now that I see there is a dedicated function already there, I will change it to use this (simpler) one.