introlab/rtabmap #1404: Repeated crashes occurred on iPhone(iPhone 13 Pro / iPhone 14 Pro)

AyushBharadwaj, 2024-12-09:
Hello @matlabbe,

I'm facing an issue on iPhone where the app crashes repeatedly while assembling a textured mesh after scanning a large amount of data. The RAM usage increases significantly, reaching around 3.0 GB, before the crash. I need a solution to handle large data scans and successfully assemble them into a textured model without causing the app to crash.

matlabbe, 2024-12-10:
I think iOS doesn't allow Apps to take [more than 5GB of RAM](https://appleinsider.com/articles/21/06/25/apple-offers-developers-workaround-for-ram-app-limits-in-ios-15-ipados-15). Depending at which step of assembling is, it may do a temporary copy of that 3GB, increasing over the limit. Maybe that copy could be avoided, but it would require some debugging and tests to investigate.

On the short term, it is possible to generate the same textured model using the [Desktop App](https://github.com/introlab/rtabmap/releases/tag/0.21.4). You can look at this [example](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#7-export-the-mesh).

matlabbe, 2024-12-14:
Here different options to use less memory during assembling. If you go under Assembling Settings:
1. you can increase Voxel Size
2. you can set "Reconstruction Depth" to 8 or 7

When clicking on Assemble action, select a lower number of polygons.