introlab/rtabmap #1410: IOS - Issue with Pose Extraction and Verification in SQLite Database

dscho15, 2024-12-11:
Hi there,

I'm working on a hobby project involving the extraction of pose, image, depth map, and camera calibration data from your application's SQLite3 database. So far, I've made significant progress:

- Successfully extracted calibration data from the "Data" table, including a sensible focal length and cx/cy coordinates
- Retrieved images and depth maps from the "Data" table
- Extracted poses and opt-indices from the "Admin" table
- Decoded poses using zlib, resulting in N x 3 x 4 pose matrices

However, I've encountered an unexpected discrepancy. When comparing the first two poses using the transformation np.linalg.inv(poses[2]) @ poses[1], the results do not align with Open3D's local adjustment algorithm's output, even when using the same initial pose calculation. It is like a couple of centimeters off... Also just to check if I was unlucky, I created a couple of different videoes (1-3 seconds each). 

So, I believe, there could be an error in my pose decoding process. 

`
def extract_opt_poses(self):
        opt_poses = zlib.decompress(opt_poses) # bytestream
        opt_poses = decode_array(opt_poses, np.float32, shape=(-1, 3, 4))
        zeros = np.zeros((len(opt_poses), 1, 4))
        zeros[:, 0, 3] = 1
        return np.hstack([opt_poses, zeros])
`

I would appreciate any insights you can provide to help understand what could be wrong (: 

Also the code base is available at https://github.com/dscho15/rtabmap-splatting

dscho15, 2024-12-11:
Update

It is not the poses. RTAB-Map seems to agree on the poses. So I guess it is either the depth-maps which are decoded wrongly, or the calibration data.

dscho15, 2024-12-12:
Gotten it to work, I will throw the code into the repository immediately :)

closed 2024-12-12 by dscho15

matlabbe, 2024-12-14:
In opt_poses, it would be the device poses. To get the camera poses, you would have to multiply the pose by the local transform contained in the calibration data (which is the transform from device->camera). That is maybe why there was a 2 cm offset.

Also, thanks for sharing your project, very interesting.