introlab/rtabmap #1411: undefined reference in android compilation

zhounao, 2024-12-12:
I follow the instructions in #1333 and have following errors:

/opt/android/arm64-v8a/include/pcl/registration/impl/registration.hpp:62: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::setInputSource(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/registration.h:199: undefined reference to `pcl::PCLBase<pcl::PointXYZ>::setInputCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `Registration':
/opt/android/arm64-v8a/include/pcl/registration/registration.h:101: undefined reference to `pcl::PCLBase<pcl::PointXYZ>::PCLBase()'
/opt/android/arm64-v8a/include/pcl/registration/registration.h:103: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
/opt/android/arm64-v8a/include/pcl/registration/registration.h:104: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::CorrespondenceEstimationBase<pcl::PointXYZ, pcl::PointXYZ, float>::setInputTarget(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/impl/correspondence_estimation.hpp:74: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::setPointRepresentation(boost::shared_ptr<pcl::PointRepresentation<pcl::PointXYZ> const> const&)'
/opt/android/arm64-v8a/include/pcl/registration/impl/correspondence_estimation.hpp:67: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::CorrespondenceEstimationBase<pcl::PointXYZ, pcl::PointXYZ, float>::setInputSource(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/correspondence_estimation.h:134: undefined reference to `pcl::PCLBase<pcl::PointXYZ>::setInputCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::IterativeClosestPoint<pcl::PointXYZ, pcl::PointXYZ, float>::computeTransformation(pcl::PointCloud<pcl::PointXYZ>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&)':
/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:196: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:212: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:238: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::IterativeClosestPoint<pcl::PointXYZ, pcl::PointXYZ, float>::determineRequiredBlobData()':
/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:261: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:265: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o):/opt/android/arm64-v8a/include/pcl/registration/impl/icp.hpp:281: more undefined references to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)' follow
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::setInputSource(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/registration.h:199: undefined reference to `pcl::PCLBase<pcl::PointXYZ>::setInputCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::setInputTarget(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/impl/registration.hpp:62: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::DefaultConvergenceCriteria<float>::hasConverged()':
/opt/android/arm64-v8a/include/pcl/registration/impl/default_convergence_criteria.hpp:51: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
/opt/android/arm64-v8a/include/pcl/registration/impl/default_convergence_criteria.hpp:70: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
/opt/android/arm64-v8a/include/pcl/registration/impl/default_convergence_criteria.hpp:89: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::TransformationEstimation2D<pcl::PointXYZ, pcl::PointXYZ, float>::estimateRigidTransformation(pcl::PointCloud<pcl::PointXYZ> const&, pcl::PointCloud<pcl::PointXYZ> const&, Eigen::Matrix<float, 4, 4, 0, 4, 4>&) const':
/opt/android/arm64-v8a/include/pcl/registration/impl/transformation_estimation_2D.hpp:50: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o):/opt/android/arm64-v8a/include/pcl/registration/impl/transformation_estimation_2D.hpp:69: more undefined references to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)' follow
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::align(pcl::PointCloud<pcl::PointXYZ>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&)':
/
/opt/android/arm64-v8a/include/pcl/registration/registration.h:199: undefined reference to `pcl::PCLBase<pcl::PointXYZINormal>::setInputCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZINormal> const> const&)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::Registration<pcl::PointXYZINormal, pcl::PointXYZINormal, float>::setInputTarget(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZINormal> const> const&)':
/opt/android/arm64-v8a/include/pcl/registration/impl/registration.hpp:62: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::TransformationEstimationSVD<pcl::PointXYZINormal, pcl::PointXYZINormal, float>::estimateRigidTransformation(pcl::PointCloud<pcl::PointXYZINormal> const&, pcl::PointCloud<pcl::PointXYZINormal> const&, Eigen::Matrix<float, 4, 4, 0, 4, 4>&) const':
/opt/android/arm64-v8a/include/pcl/registration/impl/transformation_estimation_svd.hpp:55: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::TransformationEstimationSVD<pcl::PointXYZINormal, pcl::PointXYZINormal, float>::estimateRigidTransformation(pcl::PointCloud<pcl::PointXYZINormal> const&, std::__ndk1::vector<int, std::__ndk1::allocator<int> > const&, pcl::PointCloud<pcl::PointXYZINormal> const&, Eigen::Matrix<float, 4, 4, 0, 4, 4>&) const':
/opt/android/arm64-v8a/include/pcl/registration/impl/transformation_estimation_svd.hpp:74: undefined reference to `pcl::console::print(pcl::console::VERBOSITY_LEVEL, char const*, ...)'
../../../lib/librtabmap_core.a(util3d_registration.cpp.o): In function `pcl::registration::TransformationEstimationSVD<pcl::PointXYZINormal, pcl::PointXYZINormal, float>::estimateRigidTransformation(pcl::PointCloud<pcl::PointXYZINormal> const&, std::__ndk1::vector<int, std::__ndk1::allocator<int> > const&, pcl::PointCloud<pcl::PointXYZINormal> const&, std::__ndk1::vector<int, std::__ndk1::allocator<int> > const&, Eigen::Matrix<float, 4, 4, 0, 4, 4>&) const':
[... cut ...]
app/android/jni/CMakeFiles/NativeRTABMap.dir/build.make:460: recipe for target 'app/android/libs/arm64-v8a/libNativeRTABMap.so' failed
make[2]: *** [app/android/libs/arm64-v8a/libNativeRTABMap.so] Error 1
make[2]: Leaving directory '/workspaces/rtabmap/build/arm64-v8a'
CMakeFiles/Makefile2:417: recipe for target 'app/android/jni/CMakeFiles/NativeRTABMap.dir/all' failed
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
make[1]: Leaving directory '/workspaces/rtabmap/build/arm64-v8a'
Makefile:168: recipe for target 'all' failed
make: *** [all] Error 2

I tried solution for #127 but it doesn't work.

matlabbe, 2024-12-14:
For android, I don't maintain PCL compatibility with newer versions. This is how we currently build PCL 1.8 for android:
https://github.com/introlab/rtabmap/blob/626bf6498345a5bc6fd27e61193fda0cfe075015/docker/bionic/android/deps.bash#L126-L141

If you are using a newer version, you may still try with same cmake options to see if it can help.

zhounao, 2024-12-20:
Thank you for your reply! I have changed to ubuntu18 and it works. I now have one question. How can I compile our own RGBD camera to android app? Should I directly modify code `app/android/jni/CameraARCore.h` and `app/android/jni/CameraARCore.cpp` ?

matlabbe, 2024-12-22:
I don't have example of using external camera with the android app (like using any of these [camera drivers](https://github.com/introlab/rtabmap/tree/master/corelib/src/camera) in android). The easiest could be to modify the code of one the the camera already used by the android app (like CameraARCore). Note that if you have a Java interface for your camera, you may look at how ArCore Java driver is done on Java code side, then if you send all appropriate values to c++ bridge (which uses CameraMobile driver), you may not need to modify any c++ code.