introlab/rtabmap #1417: No odometry sensor driver for Realsense?

APOS80, 2024-12-20:
Hello

I miss a driver for the built in IMU I my D455f,
Can it be added?

matlabbe, 2024-12-22:
Does D455f provide visual odometry? The Odometry Sensor under Soruce panel would be only used for sensors/camera drivers that already provide visual odometry (e.g., ZED camera or T265 camera). If you search rtabmap's odometry settings, open Advanced settings, then click on Odometry. RTAB-Map's visual odometry will be used by default when the camera drivers don't provide any odometry.

The IMU from D455f would be already published when using Realsense2 driver, unless the API is different than with D435i or D455 series.

APOS80, 2024-12-23:
Okay.

closed 2024-12-23 by APOS80