introlab/rtabmap #1420: [BUG] On iOS, RTAB-Map crashes after pressing "stop" button in data recording mode

ser94mor, 2024-12-23:
From my experience, in 95% of cases of data collection sessions in data recording mode, I observe RTAB-Map app crashes after pressing the "stop" button.

The iPhone is configured to record data at maximum frequency.

This behaviour is observed on both iPhone 15 Pro and iPhone 12 mini.

[ANOTHER BUG]
If the iPhone battery level is slightly higher than 20%, one starts a data collection session in data recording mode, and then the battery level drops below 20%, iPhone displays a pop up window suggesting to put iPhone into the Power Saving mode, which, in turn, triggers RTAB-Map app crash. Would be good if the RTAB-Map app didn't stop data recording when pop up windows appear.

matlabbe, 2024-12-26:
That recording feature is somewhat experimental, I am not surprised of these issues. At max frequency (60 Hz), I didn’t test it more than 1 min or 2. At 1 Hz, it should last minutes to hours in theory, but I didn’t test for hours. 

How long are your recordings?

For the 20% battery warning, iOS may be pushing back the app in background, turning off ARKit. I think continuing recording afterwards, assuming we have to reinitialize ARKit , would not make sense because the poses would not be continuous. The app would need however to close the current session and save it, asking the user to start another session.

ser94mor, 2024-12-27:
@matlabbe , thank you for the response.

My recordings usually last from 3 to 15 mins.

About the 20% problem:
Maybe it still makes sense to continue recording after ArKit reinitialization because you then can manually identify the moment of failure and split the recording into two removing the nodes(records) that were captured during the reinitialization period.