introlab/rtabmap #1421: Windows: Error while running slam

APOS80, 2024-12-25:
[ERROR] (2024-12-25 11:34:40) RegistrationVis.cpp:1573::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=1 Models=0 StereoModels=0 weight=0
[ERROR] (2024-12-25 11:34:40) RegistrationVis.cpp:1573::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=1 Models=0 StereoModels=0 weight=0
[ERROR] (2024-12-25 11:34:41) RegistrationVis.cpp:1573::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=3 Models=0 StereoModels=0 weight=0
[ERROR] (2024-12-25 11:34:41) RegistrationVis.cpp:1573::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=3 Models=0 StereoModels=0 weight=0

Previusly i did an internal calibration of my D455f, in the Realsense viewer theres no problem.

APOS80, 2024-12-25:
The problem disappeared for reasons unknown.

APOS80, 2024-12-25:
Tried 0.21.9 and got the same message.

[ERROR] (2024-12-25 12:31:00) RegistrationVis.cpp:1705::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=1 Models=0 StereoModels=0 weight=0
[ERROR] (2024-12-25 12:31:01) RegistrationVis.cpp:1705::rtabmap::RegistrationVis::computeTransformationImpl() Calibrated camera required. Id=4 Models=0 StereoModels=0 weight=0

APOS80, 2024-12-25:
Another thing to add.

This message comes a couple of seconds in to tha scanning.

APOS80, 2024-12-25:
I rolled back the D455f firmware one step and now things work again.

matlabbe, 2024-12-26:
We may have to upgrade realsense version used in standalone version to use latest realsense firmwares. I’ll tag as potential bug to address fir next windows release.