introlab/rtabmap #1422: Question not Issue - Gaussian Splatting

PLAN8VR, 2024-12-26:
Hi, I'm trying to export a format the will work with Gaussian splat generation software, usually this is colmap format, although bundled should work. 

Question is, how can I export a format that can be imported into colmap ?

Or, is there any more information on the bundled export format as it doesn't seem to be working for me to import into other software such as reality capture. 

Would it be possible to add a colmap export option!? Many Thai s for releasing this great package!

matlabbe, 2024-12-26:
Which Gaussian splatting software are you testing? I see Colmap output has many files, which are the minimal ones do you need? If you need only camera intrinsics, image 6dof poses and dense point cloud, we may be able to add a colmap exporter from rtabmap directly. The bundler format that rtabmap can export in is meant to be compatible at least with Meshlab, like in this example: https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab. I didn’t test with reality capture though, I’ll try it later.

PLAN8VR, 2024-12-26:
Hi, thanks for the speedy reply!

I'm trying to get the exports to work with Postshot https://www.jawset.com/

Yes, I managed to import the bundler files into meshlab, as well as cloudcompare, but in Postshot, as well as reality capture, the files get rejected. I'll need to look at logs to find out why. 

Re colmap direct export, ..... yes please!! 

Gaussian Splatying requires, ply, camera intrinsics/locations and images. 

You would make a lot of people happy if you could enable that as an export option. 

Another useful camera poses file export would be alembic. Then a lot of other softwares might be able to make use of the solved odometry. 

If you need me to test some files, please send them my way!

PLAN8VR, 2024-12-31:
+── images
│   +── image1.jpg
│   +── image2.jpg
│   +── ...
+── sparse
│   +── 0
│   │   +── cameras.bin
│   │   +── images.bin
│   │   +── points3D.bin
│   +── ...
+── dense
│   +── 0
│   │   +── images
│   │   +── sparse
│   │   +── stereo
│   │   +── fused.ply
│   │   +── meshed-poisson.ply
│   │   +── meshed-delaunay.ply
│   +── ...
+── database.db