introlab/rtabmap #1432: Windows: improve edit depth image

APOS80, 2025-01-04:
Hello there.

As I'm learning more about RTab I found that editing depth images is a bit hard.

In the window for editing depth images I would like to to able to step through images and also it would be helpful I could draw a polygon around an area and delete all sensor reading in there.

/ your devoted supporter

APOS80, 2025-01-04:
Also i would like to se the RGB and when i delete depth data it could tern gray.

matlabbe, 2025-01-13:
Hi,

If you have indeed a lot of depth images to edit, the "Edit Depth Image" tool may not be efficient for your workflow. I made the tool just to modify some images having a mirror in them, which should not be the majority of the frames. I don't know if you saw this example of usage: https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#4-optional-remove-any-mirrors-before-re-processing-the-databases, but maybe in some cases you could use the "Remove Cluster" option if you didn't try it yet (could be faster than drawing a polygon). 

Normally, we should not have to edit manually frames, if most the frames need to be edited, maybe there is an issue with the camera driver providing the images (or in ROS, a node could be in charge to filter the depth images before feeding them to rtabmap).

If someone has time to implement this, I'll be happy to review it.