introlab/rtabmap #1436: iOS : libc++abi: terminating due to uncaught exception of type UException

FrankFourth, 2025-01-14:
Hello, I have integrated the application into a Flutter project. I didn’t encounter any issues when I did this before; however, in the current project, I’m facing an error. The project launches successfully, but I get an error when I attempt to perform the first scan or view the database.

I understand why the error occurs, but I have no idea how to resolve it. The error appears on the line I indicated in the attached image.

Do you have any suggestions or advice for resolving this issue? As I mentioned, I’ve been able to run and use this project many times before, both on cross-platform and native iOS projects.

[image]

libc++abi: terminating due to uncaught exception of type UException: [FATAL] {8406105792} (2025-01-14 12:20:39.173) DBDriverSqlite3.cpp:521::executeNoResultQuery() Condition (rc == SQLITE_OK) not met! [DB error: no such column: "BRIEF/Bytes:64;BRISK/Octaves:3;BRISK/PatternScale:1;BRISK/Thresh:30;Bayes/FullPredictionUpdate:false;Bayes/PredictionLC:0.1 0.36 0.30 0.16 0.062 0.0151 0.00255 0.000324 2.5e-05 1.3e-06 4.8e-08 1.2e-09 1.9e-11 2.2e-13 1.7e-15 8.5e-18 2.9e-20 6.9e-23;Bayes/VirtualPlacePriorThr:0.9;Db/TargetVersion:;DbSqlite3/CacheSize:10000;DbSqlite3/InMemory:false;DbSqlite3/JournalMode:3;DbSqlite3/Synchronous:0;DbSqlite3/TempStore:2;FAST/CV:0;FAST/Gpu:false;FAST/GpuKeypointsRatio:0.05;FAST/GridCols:0;FAST/GridRows:0;FAST/MaxThreshold:200;FAST/MinThreshold:7;FAST/NonmaxSuppression:true;FAST/Threshold:20;FREAK/NOctaves:4;FREAK/OrientationNormalized:true;FREAK/PatternScale:22;FREAK/ScaleNormalized:true;GFTT/BlockSize:3;GFTT/Gpu:false;GFTT/K:0.04;GFTT/MinDistance:7;GFTT/QualityLevel:0.0001;GFTT/UseHarrisDetector:false;GMS/ThresholdFactor:6.0;GMS/WithRotation:false;GMS/WithScale:false;GTSAM/IncRelinearizeSkip:1;GTSAM/IncRelinearizeThreshold:0.01;GTSAM/Incremental:false;GTSAM/Optimizer:1;Grid/3D:false;Grid/CellSize:0.05;Grid/ClusterRadius:0.1;Grid/DepthDecimation:4;Grid/DepthRoiRatios:0.0 0.0 0.0 0.0;Grid/FlatObstacleDetected:true;Grid/FootprintHeight:0.0;Grid/FootprintLength:0.0;Grid/FootprintWidth:0.0;Grid/GroundIsObstacle:false;Grid/MapFrameProjection:false;Grid/MaxGroundAngle:45;Grid/MaxGroundHeight:0.0;Grid/MaxObstacleHeight:0.0;Grid/MinClusterSize:10;Grid/MinGroundHeight:0.0;Grid/NoiseFilteringMinNeighbors:5;Grid/NoiseFilteringRadius:0.0;Grid/NormalK:20;Grid/NormalsSegmentation:true;Grid/PreVoxelFiltering:true;Grid/RangeMax:5.0;Grid/RangeMin:0.0;Grid/RayTracing:false;Grid/Scan2dUnknownSpaceFilled:false;Grid/ScanDecimation:1;Grid/Sensor:1;GridGlobal/AltitudeDelta:0;GridGlobal/Eroded:false;GridGlobal/FloodFillDepth:0;GridGlobal/FootprintRadius:0.0;GridGlobal/MaxNodes:0;GridGlobal/MinSize:0.0;GridGlobal/OccupancyThr:0.5;GridGlobal/ProbClampingMax:0.971;GridGlobal/ProbClampingMin:0.1192;GridGlobal/ProbHit:0.7;GridGlobal/ProbMiss:0.4;GridGlobal/UpdateError:0.01;Icp/CCFilterOutFarthestPoints:false;Icp/CCMaxFinalRMS:0.2;Icp/CCSamplingLimit:50000;Icp/CorrespondenceRatio:0.49;Icp/DebugExportFormat:;Icp/DownsamplingStep:1;Icp/Epsilon:0.001;Icp/FiltersEnabled:3;Icp/Force4DoF:false;Icp/Iterations:10;Icp/MaxCorrespondenceDistance:0.05;Icp/MaxRotation:0.17;Icp/MaxTranslation:0.05;Icp/OutlierRatio:0.85;Icp/PMConfig:;Icp/PMMatcherEpsilon:0.0;Icp/PMMatcherIntensity:false;Icp/PMMatcherKnn:1;Icp/PointToPlane:true;Icp/PointToPlaneGroundNormalsUp:0.0;Icp/PointToPlaneK:5;Icp/PointToPlaneLowComplexityStrategy:1;Icp/PointToPlaneMinComplexity:0.02;Icp/PointToPlaneRadius:0.0;Icp/RangeMax:0;Icp/RangeMin:0;Icp/ReciprocalCorrespondences:true;Icp/Strategy:0;Icp/VoxelSize:0.05;ImuFilter/ComplementaryBiasAlpha:0.01;ImuFilter/ComplementaryDoAdpativeGain:true;ImuFilter/ComplementaryDoBiasEstimation:true;ImuFilter/ComplementaryGainAcc:0.01;ImuFilter/MadgwickGain:0.1;ImuFilter/MadgwickZeta:0.0;KAZE/Diffusivity:1;KAZE/Extended:false;KAZE/NOctaveLayers:4;KAZE/NOctaves:4;KAZE/Threshold:0.001;KAZE/Upright:false;Kp/BadSignRatio:0.5;Kp/ByteToFloat:false;Kp/DetectorStrategy:6;Kp/DictionaryPath:;Kp/FlannRebalancingFactor:2.0;Kp/GridCols:1;Kp/GridRows:1;Kp/IncrementalDictionary:true;Kp/IncrementalFlann:true;Kp/MaxDepth:0;Kp/MaxFeatures:400;Kp/MinDepth:0;Kp/NNStrategy:1;Kp/NewWordsComparedTogether:true;Kp/NndrRatio:0.8;Kp/Parallelized:false;Kp/RoiRatios:0.0 0.0 0.0 0.0;Kp/SSC:false;Kp/SubPixEps:0.02;Kp/SubPixIterations:0;Kp/SubPixWinSize:3;Kp/TfIdfLikelihoodUsed:true;Marker/CornerRefinementMethod:0;Marker/Dictionary:0;Marker/Length:0.0;Marker/MaxDepthError:0.01;Marker/MaxRange:0.0;Marker/MinRange:0.0;Marker/Priors:;Marker/PriorsVarianceAngular:0.001;Marker/PriorsVarianceLinear:0.001;Marker/VarianceAngular:0.01;Marker/VarianceLinear:0.001;Mem/BadSignaturesIgnored:false;Mem/BinDataKept:true;Mem/CompressionParallelized:false;Mem/CovOffDiagIgnored:true;Mem/DepthAsMask:true;Mem/DepthCompressionFormat:.rvl;Mem/DepthMaskFloorThr:0.0;Mem/GenerateIds:true;Mem/GlobalDescriptorStrategy:0;Mem/ImageCompressionFormat:.jpg;Mem/ImageKept:false;Mem/ImagePostDecimation:1;Mem/ImagePreDecimation:1;Mem/IncrementalMemory:true;Mem/InitWMWithAllNodes:false;Mem/IntermediateNodeDataKept:false;Mem/LaserScanDownsampleStepSize:1;Mem/LaserScanNormalK:0;Mem/LaserScanNormalRadius:0.0;Mem/LaserScanVoxelSize:0.0;Mem/LocalizationDataSaved:false;Mem/MapLabelsAdded:true;Mem/NotLinkedNodesKept:true;Mem/RawDescriptorsKept:true;Mem/RecentWmRatio:0.2;Mem/ReduceGraph:false;Mem/RehearsalIdUpdatedToNewOne:false;Mem/RehearsalSimilarity:0.3;Mem/RehearsalWeightIgnoredWhileMoving:false;Mem/RotateImagesUpsideUp:false;Mem/STMSize:10;Mem/SaveDepth16Format:false;Mem/StereoFromMotion:false;Mem/TransferSortingByWeightId:false;Mem/UseOdomFeatures:true;Mem/UseOdomGravity:true;ORB/EdgeThreshold:19;ORB/FirstLevel:0;ORB/Gpu:false;ORB/NLevels:3;ORB/PatchSize:31;ORB/ScaleFactor:2;ORB/ScoreType:0;ORB/WTA_K:2;Odom/AlignWithGround:false;Odom/Deskewing:true;Odom/FillInfoData:true;Odom/FilteringStrategy:0;Odom/GuessMotion:true;Odom/GuessSmoothingDelay:0;Odom/Holonomic:true;Odom/ImageBufferSize:1;Odom/ImageDecimation:1;Odom/KalmanMeasurementNoise:0.01;Odom/KalmanProcessNoise:0.001;Odom/KeyFrameThr:0.3;Odom/ParticleLambdaR:100;Odom/ParticleLambdaT:100;Odom/ParticleNoiseR:0.002;Odom/ParticleNoiseT:0.002;Odom/ParticleSize:400;Odom/ResetCountdown:0;Odom/ScanKeyFrameThr:0.9;Odom/Strategy:0;Odom/VisKeyFrameThr:150;OdomF2M/BundleAdjustment:1;OdomF2M/BundleAdjustmentMaxFrames:10;OdomF2M/MaxNewFeatures:0;OdomF2M/MaxSize:2000;OdomF2M/ScanMaxSize:2000;OdomF2M/ScanRange:0;OdomF2M/ScanSubtractAngle:45;OdomF2M/ScanSubtractRadius:0.05;OdomF2M/ValidDepthRatio:0.75;OdomFovis/BucketHeight:80;OdomFovis/BucketWidth:80;OdomFovis/CliqueInlierThreshold:0.1;OdomFovis/FastThreshold:20;OdomFovis/FastThresholdAdaptiveGain:0.005;OdomFovis/FeatureSearchWindow:25;OdomFovis/FeatureWindowSize:9;OdomFovis/InlierMaxReprojectionError:1.5;OdomFovis/MaxKeypointsPerBucket:25;OdomFovis/MaxMeanReprojectionError:10.0;OdomFovis/MaxPyramidLevel:3;OdomFovis/MinFeaturesForEstimate:20;OdomFovis/MinPyramidLevel:0;OdomFovis/StereoMaxDisparity:128;OdomFovis/StereoMaxDistEpipolarLine:1.5;OdomFovis/StereoMaxRefinementDisplacement:1.0;OdomFovis/StereoRequireMutualMatch:true;OdomFovis/TargetPixelsPerFeature:250;OdomFovis/UpdateTargetFeaturesWithRefined:false;OdomFovis/UseAdaptiveThreshold:true;OdomFovis/UseBucketing:true;OdomFovis/UseHomographyInitialization:true;OdomFovis/UseImageNormalization:false;OdomFovis/UseSubpixelRefinement:true;OdomLOAM/AngVar:0.01;OdomLOAM/LinVar:0.01;OdomLOAM/LocalMapping:true;OdomLOAM/Resolution:0.2;OdomLOAM/ScanPeriod:0.1;OdomLOAM/Sensor:2;OdomMSCKF/FastThreshold:10;OdomMSCKF/GridCol:5;OdomMSCKF/GridMaxFeatureNum:4;OdomMSCKF/GridMinFeatureNum:3;OdomMSCKF/GridRow:4;OdomMSCKF/InitCovAccBias:0.01;OdomMSCKF/InitCovExRot:0.00030462;OdomMSCKF/InitCovExTrans:0.000025;OdomMSCKF/InitCovGyroBias:0.01;OdomMSCKF/InitCovVel:0.25;OdomMSCKF/MaxCamStateSize:20;OdomMSCKF/MaxIteration:30;OdomMSCKF/NoiseAcc:0.05;OdomMSCKF/NoiseAccBias:0.01;OdomMSCKF/NoiseFeature:0.035;OdomMSCKF/NoiseGyro:0.005;OdomMSCKF/NoiseGyroBias:0.001;OdomMSCKF/OptTranslationThreshold:0;OdomMSCKF/PatchSize:15;OdomMSCKF/PositionStdThreshold:8.0;OdomMSCKF/PyramidLevels:3;OdomMSCKF/RansacThreshold:3;OdomMSCKF/RotationThreshold:0.2618;OdomMSCKF/StereoThreshold:5;OdomMSCKF/TrackPrecision:0.01;OdomMSCKF/TrackingRateThreshold:0.5;OdomMSCKF/TranslationThreshold:0.4;OdomMono/InitMinFlow:100;OdomMono/InitMinTranslation:0.1;OdomMono/MaxVariance:0.01;OdomMono/MinTranslation:0.02;OdomOKVIS/ConfigPath:;OdomORBSLAM/AccNoise:0.1;OdomORBSLAM/AccWalk:0.0001;OdomORBSLAM/Bf:0.076;OdomORBSLAM/Fps:0.0;OdomORBSLAM/Gy [... cut ...] s:400;Vis/MeanInliersDistance:0.0;Vis/MinDepth:0;Vis/MinInliers:25;Vis/MinInliersDistribution:0.0;Vis/PnPFlags:0;Vis/PnPMaxVariance:0.0;Vis/PnPRefineIterations:0;Vis/PnPReprojError:2;Vis/PnPSamplingPolicy:1;Vis/PnPSplitLinearCovComponents:false;Vis/PnPVarianceMedianRatio:4;Vis/RefineIterations:5;Vis/RoiRatios:0.0 0.0 0.0 0.0;Vis/SSC:false;Vis/SubPixEps:0.02;Vis/SubPixIterations:0;Vis/SubPixWinSize:3;g2o/Baseline:0.075;g2o/Optimizer:0;g2o/PixelVariance:1.0;g2o/RobustKernelDelta:8;g2o/Solver:0");]

FrankFourth, 2025-01-17:
I have re-implemented it from scratch but I still get the same error. When I tried it on a new Flutter project before, there was no problem: @matlabbe 

flutter: The Dart VM service is listening on http://127.0.0.1:58740/sBBsOflxaL0=/
10.25.0 - [FirebaseAnalytics][I-ACS800023] No pending snapshot to activate. SDK name: app_measurement
10.25.0 - [FirebaseCrashlytics][I-CLS000000] Failed to download settings. If this is your first time launching the app, make sure you have enabled Crashlytics in the Firebase Console. Error Domain=FIRCLSNetworkError Code=-5 "(null)" UserInfo={status_code=404, type=2, request_id=, content_type=text/html; charset=utf-8}
10.25.0 - [FirebaseAnalytics][I-ACS023012] Analytics collection enabled
10.25.0 - [FirebaseAnalytics][I-ACS023220] Analytics screen reporting is enabled. Call Analytics.logEvent(AnalyticsEventScreenView, parameters: [...]) to log a screen view event. To disable automatic screen reporting, set the flag FirebaseAutomaticScreenReportingEnabled to NO (boolean) in the Info.plist
10.25.0 - [FirebaseSessions][I-SES000000] [Settings] Fetching newest settings failed with error: JSONParseError("Failed to parse JSON to dictionary")
RTABMapApp::RTABMapApp()
RTABMapApp::RTABMapApp() end
[ INFO] {8406105792} (2025-01-17 16:37:25.413) RTABMapApp.cpp:1132::InitializeGLContent() 
Setting param "Rtabmap/DetectionRate"  to "1"
Setting param "Rtabmap/TimeThr"  to "0"
Setting param "Rtabmap/MemoryThr"  to "0"
Setting param "RGBD/LinearSpeedUpdate"  to "0"
Setting param "RGBD/AngularSpeedUpdate"  to "0.00"
Setting param "Rtabmap/LoopThr"  to "0.11"
Setting param "Mem/RehearsalSimilarity"  to "0.3"
Setting param "Kp/MaxFeatures"  to "400"
Setting param "Vis/MaxFeatures"  to "400"
Setting param "Vis/MinInliers"  to "25"
Setting param "RGBD/OptimizeMaxError"  to "2"
Setting param "Kp/DetectorStrategy"  to "6"
Setting param "Vis/FeatureType"  to "6"
Setting param "Mem/NotLinkedNodesKept"  to "true"
Setting param "RGBD/OptimizeFromGraphEnd"  to "true"
Setting param "RGBD/MaxOdomCacheSize"  to "10"
Setting param "Optimizer/Strategy"  to "2"
Setting param "RGBD/ProximityBySpace"  to "1"
Setting param "RGBD/MarkerDetection"  to "false"
[ INFO] {8406105792} (2025-01-17 16:37:25.459) scene.cpp:229::SetupViewPort() Setup viewport OpenGL: 1179x2556
Clearing all rendering data...
Scene::clear()
Clearing  meshes...
State: STATE_WELCOME
enableSetNeedsDisplay
databases.size() = 8
Opening database /var/mobile/Containers/Data/Application/8833C85F-4B06-44ED-954E-83A84C0C8521/Documents/rtabmap.tmp.db (inMemory=0, optimize=0, clearDatabase=1)
Erasing database "/var/mobile/Containers/Data/Application/8833C85F-4B06-44ED-954E-83A84C0C8521/Documents/rtabmap.tmp.db"...
Received RtabmapEventInit! Status=4 info=Loading database...
Initializing database...
[DEBUG] {8406105792} (2025-01-17 16:37:29.926) Rtabmap.cpp:322::init() path=/var/mobile/Containers/Data/Application/8833C85F-4B06-44ED-954E-83A84C0C8521/Documents/rtabmap.tmp.db
[ INFO] {8406105792} (2025-01-17 16:37:29.926) Rtabmap.cpp:327::init() Using database "/var/mobile/Containers/Data/Application/8833C85F-4B06-44ED-954E-83A84C0C8521/Documents/rtabmap.tmp.db".
[DEBUG] {8406105792} (2025-01-17 16:37:29.930) VWDictionary.cpp:470::update() incremental=1
[DEBUG] {8406105792} (2025-01-17 16:37:29.930) FlannIndex.cpp:52::release() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.931) FlannIndex.cpp:80::release() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.931) VWDictionary.cpp:681::update() Dictionary updated! (size=0 added=0 removed=0)
[DEBUG] {8406105792} (2025-01-17 16:37:29.931) VWDictionary.cpp:689::update() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.931) Registration.cpp:49::create() type=0
[DEBUG] {8406105792} (2025-01-17 16:37:29.951) GlobalDescriptorExtractor.cpp:51::create() Creating global descriptor of type 0
[DEBUG] {8406105792} (2025-01-17 16:37:29.952) Memory.cpp:562::parseParameters() 
Received RtabmapEventInit! Status=0 info=
[DEBUG] {8406105792} (2025-01-17 16:37:29.963) Memory.cpp:171::init() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.963) Memory.cpp:562::parseParameters() 
Received RtabmapEventInit! Status=4 info=Clearing memory...
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:1763::clear() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:6262::cleanUnusedWords() Removing 0 words (dictionary size=0)...
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:1819::clear() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:1843::clear() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:1849::clear() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:6262::cleanUnusedWords() Removing 0 words (dictionary size=0)...
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) VWDictionary.cpp:694::clear() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) FlannIndex.cpp:52::release() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) FlannIndex.cpp:80::release() 
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) Memory.cpp:1880::clear() 
Received RtabmapEventInit! Status=4 info=Clearing memory, done!
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) DBDriverSqlite3.cpp:62::DBDriverSqlite3() treadSafe=1
[DEBUG] {8406105792} (2025-01-17 16:37:29.972) DBDriverSqlite3.cpp:200::setDbInMemory() dbInMemory=0
Received RtabmapEventInit! Status=4 info=Connecting to database "/var/mobile/Containers/Data/Application/8833C85F-4B06-44ED-954E-83A84C0C8521/Documents/rtabmap.tmp.db"...
[... cut ...] s:400;Vis/MeanInliersDistance:0.0;Vis/MinDepth:0;Vis/MinInliers:25;Vis/MinInliersDistribution:0.0;Vis/PnPFlags:0;Vis/PnPMaxVariance:0.0;Vis/PnPRefineIterations:0;Vis/PnPReprojError:2;Vis/PnPSamplingPolicy:1;Vis/PnPSplitLinearCovComponents:false;Vis/PnPVarianceMedianRatio:4;Vis/RefineIterations:5;Vis/RoiRatios:0.0 0.0 0.0 0.0;Vis/SSC:false;Vis/SubPixEps:0.02;Vis/SubPixIterations:0;Vis/SubPixWinSize:3;g2o/Baseline:0.075;g2o/Optimizer:0;g2o/PixelVariance:1.0;g2o/RobustKernelDelta:8;g2o/Solver:0");]

matlabbe, 2025-01-19:
The error boiled down to:
`
[FATAL] {8406105792} (2025-01-17 16:37:29.998) DBDriverSqlite3.cpp:521::executeNoResultQuery() Condition (rc == SQLITE_OK) not met! [DB error: no such column: "PARAM/KEY:VALUE;PARAM/KEY:VALUE" - should this be a string literal in single-quotes?, the query is INSERT INTO Info(STM_size,last_sign_added,process_mem_used,database_mem_used,dictionary_size,parameters) values(0,0,0,0,0,"PARAM/KEY:VALUE;PARAM/KEY:VALUE");]
`

Maybe this "should this be a string literal in single-quotes?" could give an cue on how to fix this. We would maybe need to change the double quotes here by single quotes:
https://github.com/introlab/rtabmap/blob/3ab9a69fbb31033b618f01bd2897401ce4957044/corelib/src/DBDriver.cpp#L1182

It seems sqlite3 can support both single and double quotes for literal string, but the standard SQL they are different:
https://www.sqlite.org/quirks.html#double_quoted_string_literals_are_accepted

We could change all similar double quotes to single quotes if it solves your issue. On sqlite3, there should be no impact.

FrankFourth, 2025-01-20:
Thank you very much for your answer. However, I am confused. Why does this problem occur? In other words, it does not cause any problems when integrated into a Flutter project created from scratch, but it causes problems in an existing project. @matlabbe

matlabbe, 2025-02-01:
Different platform architecture and/or compilation/project parameters and/or compiler version and/or different sql library version used. Anyway, I changed it to single quoted string in this commit https://github.com/introlab/rtabmap/commit/fcfadfe48937f46874bc70b429b99fb22eccb121

closed 2025-02-01 by matlabbe