introlab/rtabmap #1438: Occupancy map/grid not updating in "mapping" mode.

sandeepdutta, 2025-01-19:
Hi Matthew,

I updated to "HEAD" on master & noticed the following problem.
I created a new map using , saved in rtabmap.db, then I restart 
with same parameters the occupancy grid is no longer updated,
the database grows larger, but when I open the database with 
rtabmap-databaseviewer it does not show the new nodes.
I reverted back to "commit 626bf6498345a5bc6fd27e61193fda0cfe075015".
and it started working again, not sure if this a database problem
or something new that I should add new parameters for.

I will be happy to help you debug if you have some suggestions.
Thank you 

Sandeep

matlabbe, 2025-01-19:
There are no related changes from the version of Dec 14 that could cause such issue. If you can try other recent versions and find which commit is not working, that could help to know where to look at.

sandeepdutta, 2025-01-20:
After clean build of both rtabmap & rtabmap_ros this issue went away.

closed 2025-01-20 by sandeepdutta