introlab/rtabmap #1440: Z-axis drift in odom VINS

xxxy12, 2025-01-22:
I want to use VINS as odom but find it always have drift in Z-axis and only in Z-axis, which is not occur to f2m(I move my robot in 2D ground).

The result of vins like:
[image]
[image]

This cause my floor split:
[image]

Compare with f2m as follows:
[image]
[image]

Does this caused by the opticalFlow method in feature point extraction used by vins? But why the drift only appear in Z-axis? How can I solve it? Looking forward to your response.

xxxy12, 2025-01-23:
Here is my [rtabmap_vins.db](https://github.com/xxxy12/myRtabmap)

matlabbe, 2025-02-03:
It is difficult to say. The map is actually tilted ~30 degs, so the drift appears on all axes. F2M would probably do the same long-term. If the camera is always moving at the same height, fixed at same angle and the environment is mostly 2D, so that you could expect odometry to be always 2D, you can enable `Reg/Force3DoF` to remove completely the drift in Z.