introlab/rtabmap #1444: Failing Localization with Obstructions Present while still

jsreveal, 2025-01-29:
I am experiencing difficulties with RTAB-Map's localization and loop closure capabilities in scenarios where there are obstructions (specifically, people) between the camera and previously mapped areas, but there are still important scene features visible. In this scenario the camera subject is still. I don't experience these issues if the people are not obstructing the view. 

**Observations:**

- **Failure Modes:** RTAB-Map fails to either localize correctly or establish loop closures when obstructions are present in the described scenario. Instead it yields bad loop closures.
-
- **External SIFT Verification**: I have independently tested SIFT feature detection and matching using the same configuration as RTAB-Map's internal implementation (see below). When comparing RGB images from the current scene (with people present) against previously captured map images, SIFT successfully identifies good keypoint matches, not many but feels sufficien. This suggests that the features themselves are detectable and match-able despite the obstructions.
- **Obstruction Removal:** When the people are removed from the scene, RTAB-Map is able to localize correctly without issue.

**Scenario with occlusion, with bad localization**
There are some good keypoint matches, despite some bad ones.
Image on the right is an rgb exported from the map at a close position. Using distance threshold 0.7
[image]

**Scenario without occlusion  with good localization** 
Good keypoint matches, but again not many.Image on the left is from an rgb exported from the map
[image]

**Expected Behavior:**

I would expect RTAB-Map to be more robust to partial occlusions, especially when feature matching (as demonstrated by external SIFT) is still possible. Ideally, it should be able to utilize the available matches to achieve at least a coarse localization or loop closure, even if the accuracy is slightly reduced due to the obstructions.

**Question:**

What factors within RTAB-Map's processing could be contributing to these failures in the presence of partial occlusions, despite the apparent availability of matching features? Are there specific parameters or configurations that could be adjusted to improve robustness in such scenarios?

matlabbe, 2025-02-04:
Are you using a stereo or RGB-D camera? If a depth image is used, by default features are not extracted when there is no depth. You may disable this by setting `Mem/DepthAsMask: false` and `Vis/DepthAsMask: false`. The motion estimation would be accepted only if the number of inliers is above 20 by default, it can be changed with parameter `Vis/MinInliers`. Also make sure rtabmap is using SIFT: `Kp/DetectorStrategy: 1` and `Mem/UseOdomFeatures: false`. 

Note that another way to debug localization is to backup your map database, then restart in localization mode with `Mem/LocalizationDataSaved: true`. Do the the localization session, then close. The database will contain the data recorded during the localization session. Open the database in rtabmap-databaseViewer, then you can browse one map frame and one localization frame, then with Constraint View, check the checkbox to enable adding loop closures and click Add. You can enable Debug log under Gui settings panel to look at why a new added constraint would fail. It is also possible to change the Core Parameters and re-do Add to see how the parameters affect loop closure transform estimation. Enable `RGBD/LoopClosureReextractFeatures` under Core parameters to force re-extracting features depending on the current strategy used to test other features for example.

jsreveal, 2025-02-11:
I am using monocular localization. Those configuration changes did not produce better results. 

Also tried to use superpoint+superglue but without improvements.

After doing localization with save localization data, I cannot open the database from databaseViewer. 

[image]

Thank you,
João

jsreveal, 2025-02-12:
Ok, so I have set the location images with a dummy pose and was able to open the databse in rtabmap-databaseViewer with the location images.

I have ran location, against a map created with superpoint and superglue (complete config end of the comment). The matches I can see in rtabmap-databaseViewer are much less than the onews I obtain when I run superpoint+superglue from the superglue repository. 
**Does location run superglue at all?**

Matches shown in databaseViewer:
[image]

Matches running superpoint+superglue using demo_superglue.py (config identical):

[image]

map config:
Parameters (Yellow=modified, Red=old parameter not used anymore, NA=not in database):
BRIEF/Bytes=                       32
BRISK/Octaves=                     3
BRISK/PatternScale=                1
BRISK/Thresh=                      30
Bayes/FullPredictionUpdate=        false
Bayes/PredictionLC=                0.1 0.36 0.30 0.16 0.062 0.0151 0.00255 0.000324 2.5e-05 1.3e-06 4.8e-08 1.2e-09 1.9e-11 2.2e-13 1.7e-15 8.5e-18 2.9e-20 6.9e-23
Bayes/VirtualPlacePriorThr=        0.9
DbSqlite3/CacheSize=               10000
DbSqlite3/InMemory=                false
DbSqlite3/JournalMode=             3
DbSqlite3/Synchronous=             0
DbSqlite3/TempStore=               2
FAST/CV=                           0
FAST/Gpu=                          false
FAST/GpuKeypointsRatio=            0.05
FAST/GridCols=                     0
FAST/GridRows=                     0
FAST/MaxThreshold=                 200
FAST/MinThreshold=                 7
FAST/NonmaxSuppression=            true
FAST/Threshold=                    20
FREAK/NOctaves=                    4
FREAK/OrientationNormalized=       true
FREAK/PatternScale=                22
FREAK/ScaleNormalized=             true
GFTT/BlockSize=                    3
GFTT/Gpu=                          false
GFTT/K=                            0.04
GFTT/MinDistance=                  7
GFTT/QualityLevel=                 0.001
GFTT/UseHarrisDetector=            false
GMS/ThresholdFactor=               6
GMS/WithRotation=                  false
GMS/WithScale=                     false
GTSAM/IncRelinearizeSkip=          1
GTSAM/IncRelinearizeThreshold=     0.01
GTSAM/Incremental=                 false
GTSAM/Optimizer=                   1
Grid/3D=                           true
Grid/CellSize=                     0.05
Grid/ClusterRadius=                0.1
Grid/DepthDecimation=              4
Grid/DepthRoiRatios=               0.0 0.0 0.0 0.0
Grid/FlatObstacleDetected=         true
Grid/FootprintHeight=              0.0
Grid/FootprintLength=              0.0
Grid/FootprintWidth=               0.0
Grid/GroundIsObstacle=             false
Grid/MapFrameProjection=           false
Grid/MaxGroundAngle=               45
Grid/MaxGroundHeight=              0.0
Grid/MaxObstacleHeight=            0.0
Grid/MinClusterSize=               10
Grid/MinGroundHeight=              0
Grid/NoiseFilteringMinNeighbors=   5
Grid/NoiseFilteringRadius=         0.0
[... 485 lines cut ...]
Vis/PnPVarianceMedianRatio=        4
Vis/RefineIterations=              5
Vis/RoiRatios=                     0.0 0.0 0.0 0.0
Vis/SSC=                           false
Vis/SubPixEps=                     0.02
Vis/SubPixIterations=              0
Vis/SubPixWinSize=                 3
g2o/Baseline=                      0.075
g2o/Optimizer=                     0
g2o/PixelVariance=                 1
g2o/RobustKernelDelta=             8
g2o/Solver=                        0
Db/TargetVersion=                  NA  (default="")
Icp/DebugExportFormat=             NA  (default="")
Icp/PMConfig=                      NA  (default="")
Kp/DictionaryPath=                 NA  (default="")
Marker/Priors=                     NA  (default="")
OdomOKVIS/ConfigPath=              NA  (default="")
OdomORBSLAM/VocPath=               NA  (default="")
OdomOpenVINS/LeftMaskPath=         NA  (default="")
OdomOpenVINS/RightMaskPath=        NA  (default="")
OdomVINS/ConfigPath=               NA  (default="")
PyDescriptor/Path=                 NA  (default="")
PyDetector/Path=                   NA  (default="")
Rtabmap/WorkingDirectory=          NA  (default="")

matlabbe, 2025-03-01:
To verify that superglue is giving similar results on both cases, you can try `rtabmap-matcher` tool that can be used to test the matching algorithms.
```
rtabmap-matcher --help
...
rtabmap-matcher \
   --Vis/FeatureType 11 \
   --SuperPoint/ModelPath "superpoint.pt" \
   --Vis/CorNNType 6 \
   --PyMatcher/Path "~/SuperGluePretrainedNetwork/rtabmap_superglue.py" \
   from.png to.png
...
``` 
You can post the log here with `--udebug` for more help.