introlab/rtabmap #1449: How to extract raw stereo images(left and right camera images) from database created by Rtabmap.

skitmanish, 2025-02-10:
How can I retrieve raw stereo images (left and right camera images) from a database created by RTAB-Map? Does the database contain a field that stores left and right camera images?

matlabbe, 2025-02-12:
You can use `rtabmap-databaseViewer` and do File -> Extract images...

You can also use `rtabmap-export --images rtabmap.db`.

If you want to extract yourself from the database, they are compressed blob that can be decoded by opencv's `imdecode` function:
https://github.com/introlab/rtabmap/blob/618e839999f5717f5269c6dff5e4868e6ad9d045/corelib/src/resources/DatabaseSchema.sql.in#L33-L34
(depth is the right image in a stereo config)

matlabbe, 2025-03-01:
Don't provide decimation to keep same resolution. To keep the right image as rgb, you may have to hard-code the output encoding here to `bgr8` instead of `mono8`:
https://github.com/introlab/rtabmap_ros/blob/a57b2971f2079d375f867bdb6927efefd75ddd38/rtabmap_conversions/src/MsgConversion.cpp#L2494
However, I am not totally sure at 100% that it would not cause any issue.

matlabbe, 2025-03-09:
Added a PR to support right color image downstream: https://github.com/introlab/rtabmap/pull/1463