introlab/rtabmap #1453: Adding custom odometry to RTABMAP

Kavi91, 2025-02-20:
@matlabbe Sir, Please accept my sincere gratitude for implementing RTABMAP and making it open source for future developments.

So, I am planning to add my version of deep learning-based odometry into RTABMAP. This odometry mainly fused the camera and lidar together to produce a single odometry. Features will also extracted using Deep neural networks for both lidar and RGB.

In order to implement this with current framework.

1. Where I should place my new files

2. which files I should modify to enable custom feature extractions and odometry ?

Thank you in advance.

Kavi91, 2025-03-01:
> [@matlabbe](https://github.com/matlabbe) Sir, Please accept my sincere gratitude for implementing RTABMAP and making it open source for future developments.
>
> So, I am planning to add my version of [...]

matlabbe, 2025-03-05:
If you are using ROS, see comment here: https://github.com/introlab/rtabmap_ros/issues/1281

For the standalone version, that is more complicated. Basically, it would be to add a new class `Odometry****` along those https://github.com/introlab/rtabmap/tree/master/corelib/src/odometry and override `Odometry::computeTransform()` abstract function. The SensorData object provided as argument would contain lidar and camera images + calibrations.

Kavi91, 2025-03-31:
@matlabbe  Thank you so much. I will try  this.