introlab/rtabmap #1455: iOS: marker detection broken?

torjusti, 2025-02-24:
Hi,

After one of the latest iOS RTAB-MAP updates I'm no longer seeing any indication that markers are being detected. Was the toast message removed or are there now other settings required to make it work?

Specifically I'm using 10 cm large APRILTAG markers from the 36h11 family, but I've tried others as well (though scanning a computer display and not a print-out, in this case) and I struggle to see any indication that they are detected with tags that previously have been detected file. I have the depth error estimation at 4 cm and the marker size hardcoded to 10 cm.

I also don't see them being detected when reprocessing a scan afterwards. I feel like this also used to state when it detected a marker?

Thanks

matlabbe, 2025-02-25:
Just tried the latest version on app store:

[image]

Make sure it is not ID=0, only IDs >=1 work with this app. A frame should appear when first detected like in screenshot above, then when re-detecting it, the background would turn "orange" and a toast should appear and says "Landmark # detected!". Doublecheck the apriltag family is the right one in the settings.

I tested with one of these tags directly on my monitor: https://introlab.3it.usherbrooke.ca/images/2/23/Tags1-4.pdf 

EDIT: I think I see what could be the issue. Whether we set a known size of the marker or not, we still need to look at the tag perpendicularly to initialize the first detection. Also, the detection seems not working at all without the lidar. I'll check to fix this in next release, https://github.com/introlab/rtabmap/pull/1451

torjusti, 2025-02-26:
Thanks, interesting. I'm having no luck still :)

I can confirm I'm trying with several markers with ID > 0. I recall seeing the toasts appear all the time before, but lately they have disappeared completely. Varying the angle and distance from the marker has no effect unfortunately, and the linked tags don't work. I have enabled the LiDAR.

If you can confirm that marker detections are also logged when reprocessing I'm happy to record a database and upload it.

matlabbe, 2025-03-10:
If they are not detected, they are not added to the database. However, we can reprocess the database so that the tags can be detected by the desktop app. 

I've found a bug that if we "Restore All Default Settings", any changes in settings will be ignored until the app is restarted. That's why it was not working in my case above. Not sure if it is your case. The workaround with the current version is that if you click on "Restore All Default Settings", close completly the app and re-open, then any changes in settings will be correctly applied. This is fixed in commit above.

matlabbe, 2025-03-10:
Another bug fixed above. The automatic marker size estimation was not working correctly. The workawound with current version is to set a known marker size.

closed 2025-03-10 by matlabbe

matlabbe, 2025-03-15:
The new version with these fixes has been released to App Store.