introlab/rtabmap #1459: Restart Localization from another source of odometry

dyumanaditya, 2025-03-01:
Hello,

I'm using rtabmap rgbd-odometry on a quadruped robot. Large and sudden movements are quite common and the robot gets lost quite easily. 

I have another source of odometry but which is much less reliable. I want to be able to restart the rtabmap rgbd odometry with this source whenever the robot gets lost, instead of having the entire system break down. How can I do this?

I have not seen any parameters that might allow me to do this, so any pointers to the code which I could potentially modify to get this to work would be very helpful as well.

Thank you in advance

matlabbe, 2025-03-05:
Use `guess_frame_id` using the TF frame of the other source of odometry. For example, we can have odom->base_link published by some wheel odometry. For rgbd_odometry, we can set `guess_frame_id:=odom` with `odom_frame_id:=vo` (may remap also `odom` topic of rgbd_odometry if it has the same name than wheel odom topic). rgbd_odometry will then publish TF vo->odom. We can set `Odom/ResetCountdown` to `1` to immediately reset vo based on latest motion computed from odom.

You can set `publish_null_when_lost:=false` to avoid publish vo when it cannot be computed. The TF vo->odom ill still be published with latest vo correction to not break the Tf tree.

dyumanaditya, 2025-03-08:
Perfect! Thank you @matlabbe!

closed 2025-03-08 by dyumanaditya