introlab/rtabmap #1465: Graph Optimization Reverts Added or Cleaned Obstacles in the Occupancy Grid Map

anath93, 2025-03-10:
Hi Mathieu,

Sorry to duplicate this from your forum but this can be confirmed bug and just been causing issue for long time and did not had chance to dive deeper into this. I know you are very busy, can you please shed light on this when you get a chance. I do not want this to go unnoticed as this will cause issues who ever is trying to use tags.

https://easyupload.io/lr5r9e

https://easyupload.io/6fspe0

[image]

This is the log screenshot where after seeing tags, which does not contain pose info gets cleared with all other poses and added again from cache where the issue comes. Meaning the changes we had made offline in the editor reverts back to map which was generated during mapping and does not contain the newly edited changes.

Below are my launch files and database.

https://easyupload.io/m/98662i

Currently I applied fix to just clear the Nodes which do not have Pose information and leave the rest alone, which works. In long run I am not sure if this can be issue !

bool GlobalMap::update(const std::map<int, Transform> & poses)
{
        UDEBUG("Update (poses=%d addedNodes_=%d)", (int)poses.size(), (int)addedNodes_.size());

        // First, check of the graph has changed. If so, re-create the octree by moving all occupied nodes.
        bool graphOptimized = false; // If a loop closure happened (e.g., poses are modified)
        bool graphChanged = addedNodes_.size()>0; // If the new map doesn't have any node from the previous map
        float updateErrorSqrd = updateError_*updateError_;
       
        for(std::map<int, Transform>::iterator iter=addedNodes_.begin(); iter!=addedNodes_.end(); ++iter)
        {
                std::map<int, Transform>::const_iterator jter = poses.find(iter->first);
                if(jter != poses.end())
                {
                        graphChanged = false;
                        UASSERT(!iter->second.isNull() && !jter->second.isNull());
                        if(iter->second.getDistanceSquared(jter->second) > updateErrorSqrd)
                        {
                                graphOptimized = true;
                        }
                }
                else
                {
                        extraNodesToClear_.insert(iter->first);  //added this to the .h file which is extranodes which we get while iteration above which do not have the pose.

                        UDEBUG("Updated pose for node %d is not found, some points may not be copied. Use negative ids to just update cell values without adding new ones.", iter->first);
                }
        }

        if(graphOptimized || graphChanged)
        {
                // clear all but keep cache
                clear();
        }

        std::list<std::pair<int, Transform> > orderedPoses;

        // add old poses that were not in the current map (they were just retrieved from LTM)
        for(std::map<int, Transform>::const_iterator iter=poses.lower_bound(1); iter!=poses.end(); ++iter)
        {
                if(!isNodeAssembled(iter->first))
                {
                        UDEBUG("Pose %d not found in current added poses, it will be added to map", iter->first);
                        orderedPoses.push_back(*iter);
                }
        }

        // insert zero after
        if(poses.find(0) != poses.end())
        {
                orderedPoses.push_back(std::make_pair(-1, poses.at(0)));
        }

        if(!orderedPoses.empty())
        {
                assemble(orderedPoses);
        }

        return !orderedPoses.empty();
}

then clear fucntion as

void GlobalMap::clear()
{
        UDEBUG("Clearing");
        //addedNodes_.clear();
        extraNodesToClear_.clear();
        minValues_[0] = minValues_[1] = minValues_[2] = 0.0;
        maxValues_[0] = maxValues_[1] = maxValues_[2] = 0.0;
}

matlabbe, 2025-03-10:
Thank you for the details, I saw the other posts in my mailbox, but I have still a 2 weeks backlog of emails to read. I want to get this one https://github.com/introlab/rtabmap/pull/1451 released soon, I'll check this issue afterwards.

anath93, 2025-03-10:
Thank you Mathieu, I appreciate you !

matlabbe, 2025-03-16:
I see in the logs that memory management is enabled (`Rtabmap/TimeThr>0`) and some nodes were transferred between LTM and WM. If either `graphOptimized` or `graphChanged` are true, we have to re-optimize the whole map.

A workaround could be to start the localization session with all nodes loaded and memory management disabled:
```
--Mem/InitWMWithAllNodes true
--Rtabmap/TimeThr 0
--Rtabmap/MemoryThr 0
```
If the optimized map was generated in rtabmap-databaseViewer with all nodes from LTM, then it may not have to be re-optimized anymore, so you won't lose your manual changes in the optimized occupancy grid.

About the tags and memory management, here some other references:
* http://official-rtab-map-forum.206.s1.nabble.com/Integrate-external-markers-from-apriltag-ros-with-Priors-td10358.html#a10454
* https://github.com/introlab/rtabmap/issues/1338

anath93, 2025-03-16:
Good Morning Mathieu, 

Thank you for the reply, I read your references and makes sense to add Landmarks as Virtual link to old node before pruning them and optimizing the map right away as the changes in the call of Optimized graph will still revert back the changes to the stock map when it is assembled. Is there a way to ensure that these updates persist in the final assembled map, from Occupancy grid cache maybe ?

I also did try disabling memory management but result was same over time, map reverts back. I did not log that and attach it just yet. 
This will just defeat the purpose of mapping in dynamic environment and using tags as the graph will revert back someday causing upstream Nav2 to act differently because unless we go in append mode of mapping, the voxel layer should take care of those dynamic objects in the static layer.
What I am thinking is, this is more of manual work, we know what nodes we deleted through the Debug logs when graph optimization is called, what if we through export DB and delete those rows using SQL commands to delete them specifically and load the DB again in localization mode ? If its possible, can you please advise how to in basic way meaning using rtabmap tools and I will write the script and post it here so if someone wants it, can also use it in future and if they want to append to this map they can call mapping_mode service which should still work. (As we can see it goes to 1 -> 28 and so forth.)

[image]

anath93, 2025-03-19:
Hi Mathieu,

I tried deleting those nodes with steps above but still that part of map does not get deleted. Can you please provide a fix, I am happy to work on it too with your direction as you know in and out of the entire library better than anyone.

matlabbe, 2025-03-19:
I think the best would be to describe your current workflow so that I can reproduce the issue here. From your posts, it seems:

### Scenario A
1. Do a mapping session with Memory Management (MM) enabled and with tag detection
2. Cleanup the **already saved** optimized map using rtabmap-databaseViewer
3. Restart in localization mode: I think at this point because MM was enabled during mapping, it is likely that some nodes will be in Long-Term Memory (LTM) and retrieved to Working Memory (WM), causing a re-generation of the map. I think in this scenario even by disabling MM and enabling `Mem/InitWMWithAllNodes`, it will cause the map to be regenerated on start because the saved optimized graph would not have all the nodes.

### Scenario B
Here another scenario:
1. Do a mapping session with Memory Management (MM) enabled and with tag detection
2. **Renegerate/overwrite** the optimized map using rtabmap-databaseViewer with **ALL** nodes
3. Cleanup the regenerated optimized map
4. Restart in localization mode with MM disabled and `Mem/InitWMWithAllNodes` enabled: **I would expect that the map will keep the changes we did** (e.g. removing all dynamic obstacles from the occupancy grid). However, if the database contains a session that was not included in the optimized graph and the robot localizes in it, it will clear the optimized map to generate one of the other session.
5. If we re-do a new mapping session to append a new part of the environment to the map or just update one part, as soon as we set back `Mem/IncrementalMemory` to true, **the map will be regenerated**.

It seems in your comments that you would want 5 to not re-generate the map and just append the new stuff. We don't have a way to update only a part of the already optimized map. The reason is that any loop closures can cause the whole graph to deform, and make some cells wrongly set to empty previously to be obstacles or vice-versa. In the past we experimented with changing only part of the map by tracking changes on every cells (`GridGlobal/FullUpdate` parameter was removed in this PR https://github.com/introlab/rtabmap/pull/1180) but it was not trivial for some edge cases.

If your use-case matches one of the scenarios above and if the results you have are different than what I described that I am expecting, I could try to reproduce that scenario in simulation to see which step is not following my expectation.

Also, are you using the remapping as a strategy to clear/add dynamic obstacles? If so, I would recommend to cleanup the optimized map once from the current dynamic obstacles (Scenario B1-4), then never remap again and keep that map as the basic static map for general planning. Then use Global Costmap plugin from nav2 to track/clear dynamic obstacles for you.

anath93, 2025-03-19:
Hi Mathieu,

Thank you for the reply, my current workflow is as follows. We do mapping with MM disabled, all we use is Rtabmap/DetectionRate to 2 Hz to accumulate as many nodes as possible instead of setting it at 0 hz, 

[image]

[image]

MM disabled In Localization:

Later in Localization mode we enable the MM and revert the detection to be 1 Hz, I have tried disabling the MM also with this setup but somewhere after we see the tag 2 or 3 times during overnight along the line find loop closure the graph gets optimized which is tricky to point exactly when it happened but it does. Note: Graph does not revert back when we launch initially. 

MM enabled In Localization:
Same as above but here its obvious when we see tag and loop closure happens, the graph gets optimized right away and reverts back the changes.  Note: it does not revert back the graph when we launch initially. 

Please correct me if we have to use MM during mapping also, as I was not aware of that. 
[Current Launch File and DB ](https://easyupload.io/m/p9skn3)

> It seems in your comments that you would want 5 to not re-generate the map and just append the new stuff. We don't have a way to update only a part of the already optimized map. The reason is that [...]

Sorry if I was not clear, but I do not want anything to append as I understand until we find global loop closure with the previous map we wont be able to append anything and the graph might deform as you suggested either ways which is not we want, as that will change the graph node locations too. 

> Also, are you using the remapping as a strategy to clear/add dynamic obstacles? If so, I would recommend to cleanup the optimized map once from the current dynamic obstacles (Scenario B1-4), then [...]

This is my last resort to come on Sunday and redo the points for Nav2 and do the entire mapping, if there is no quick patch to fix or any error on my end as well. The thing is the boundary we create in Rtabmap-Databaseviewer also reverts back as the graph is optimized and does not retain but Nav2 Keep out filter does the job as backup which is where we need the most.

matlabbe, 2025-03-20:
Just noticed that you have Marker's priors. I'll try a scenario with theses. 

For MM, not sure why you need MM in localization mode.

In your launch file `Rtabmap/TimeThr` is commented and there is no `Rtabmap/MemoryThr`, so MM seems not used either in mapping or localization modes, so the scenarios that there are nodes retrieved causing a re-optimization should not happen.

Looking at your database, it looks having all nodes in the optimized graph already + 4 landmarks.
```
Global graph:       1216 poses and 3734 links
Optimized graph:    1220 poses (x=3734->-26, y=4->-41, z=0->0)
```
I have a better idea what to test now, if there is indeed a bug, it should be fairly easy to reproduce.

anath93, 2025-03-20:
Hi Mathieu,

in this case please correct me when you can, 

> For MM, not sure why you need MM in localization mode.

My understanding is, MM helps in keeping the WM small and can result in faster pose correction as we are not searching all 1216 poses in Localization mode and we hit brakes using Rtabmap/TimeThr and I was using in combination to this was Mem/RehearsalSimilarity which is also commented out in the launch file as Rtabmap/MemoryThr is more platform based parameter and can get different results so did not thought of using it. But please shed light on what should be the right use case if I am wrong here. 
After this issue, I am also planning to port over to Composable Nodes as well too !

> I have a better idea what to test now, if there is indeed a bug, it should be fairly easy to reproduce.

Thank you !

matlabbe, 2025-03-23:
For the original issue, I cannot reproduce the map re-generated on loop closure detection or landmark detection in localization mode:
[image]
[image]

Tried with and without priors. However, I've found a way to reset the map, by changing a parameter. On update parameters callback, the map manager was resetting the maps, causing a re-generation. I fixed this in this commit https://github.com/introlab/rtabmap_ros/commit/70ddb3fc839394a5b440fee56f1877c508d6b9ea so that in localization mode, the map is not cleared when we change a parameter live.

ICP odometry should use roughly same CPU usage, it is bounded to maximum local scan map size. It may "increase" if the system is starving or is impacted by other processes using more CPU over time.

I don't see big issues in the database. The LiDAR odometry looks quite accurate, so at 2 Hz, you may be able to increase `Mem/STMSize` to 60 (currently 30) to avoid unnecessary proximity links. There is a missed loop closure at the end, but I reprocessed it again and it can be detected after adjusting correctly the odometry angular covariance. I saw that the odometry angular covariance is pretty high (`0.005`), you can try with `odom_tf_angular_variance: 0.00007569` instead. The angular error will be smaller on odometry than the landmark detection. The camera cloud and the lidar doesn't match perfectly, so adding landmarks seem to deform more the map than it should if we don't reduce odometry angular variance.
[image]

I also noticed that some marker detections have very wrong orientation estimation, but if we limit the range of the detections, it looks better. Here is the command I tested:
```
rtabmap-reprocess \
   --Mem/UseOdomFeatures true \
   --Mem/STMSize 60 \
   --Marker/MaxRange 1.2 \
   -nolandmark \
   --uwarn  \
   rtabmap.db  output.db
````

[image]

anath93, 2025-03-23:
Thank you Mathieu for taking time, I appreciate your time and response. I will make the changes and apply the commit. For now I will close the issue and reopen as the republishing of map makes sense and I see the commit as well. Also I ported over to Composable nodes (serialization/deserialization helped a lot), something to note is that CPU usage has drastically gone down to 30%.

closed 2025-03-23 by anath93