introlab/rtabmap #1467: rtabmap-reprocess restarts opt_graph counter

hkortier, 2025-03-13:
I'm reprocessing a data-recording from the ios app using the following settings and commandline script 'rtabmap-reprocess':

```
RTABMAP_SETTINGS=(
  --Rtabmap/DetectionRate 1
  --Mem/RehearsalSimilarity 0.3
  --Kp/MaxFeatures 500
  --Rtabmap/MemoryThr 0
  --RGBD/MarkerDetection false
  --RGBD/LinearUpdate 0.05
  --RGBD/AngularUpdate 0.05
)

```

Unfortunately it seems that the optimized graphs is restarted somehow:

```
Processed 955/1640 nodes [id=955 map=0 opt_graph=940]... 39ms
Processed 956/1640 nodes [id=956 map=0 opt_graph=941]... 37ms
Processed 957/1640 nodes [id=957 map=0 opt_graph=23]... 53ms Loop on 956 [0]
Processed 958/1640 nodes [id=958 map=0 opt_graph=24]... 42ms
```

Eventually I'm missing a lot of optimized poses. Can you explain this behaviour?

the database can be found [here](https://drive.google.com/file/d/1FDadyVbpVAsbF4-S2Vyaw9UUMzEyG6GR/view?usp=share_link)

hkortier, 2025-03-13:
I'm using the latest [commit](https://github.com/introlab/rtabmap/commits/master/) for both the iOS app and command line tools.

matlabbe, 2025-03-15:
rtabmap-reprocess ignore Rtabmap/DetectionRate when reprocessing the database. By default it always processes all frames. The skip frames, you should use option `-skip`.

The issue here is that the database seems to have been recorded in "Data Recording" mode, meaning that the `Mem/STMSize` was set to 1. Combining with `--RGBD/LinearUpdate 0.05 --RGBD/AngularUpdate 0.05`, as soon as one node is ignored, the STM becomes empty, then the issue happen. The map is split silently. I think there should be an error message shown when we are not in data recorder mode.

The quick fix is to set back `Mem/STMSize` to its default (10):
```
rtabmap-reprocess \
   --Rtabmap/DetectionRate 1 \
   --Mem/RehearsalSimilarity 0.3 \
   --Kp/MaxFeatures 500 \
   --Rtabmap/MemoryThr 0 \
   --RGBD/MarkerDetection false \
   --RGBD/LinearUpdate 0.05 \
   --RGBD/AngularUpdate 0.05 \
   --Mem/STMSize 10 \
   rtabmap_db_reprocess_issue.db output.db
```

However, the recorded databases in Recorder mode were meant to be reprocessed in standalone version as a database input source (like a camera), while we use default parameters when creating a new map from that data source. With `rtabmap-reprocess`, we can do something similar by simply using `-default` option to ignore parameters from that kind of database:

```
rtabmap-reprocess -default rtabmap_db_reprocess_issue.db output.db
```
That solves implicitly the `Mem/STMSize` issue.

Some comments on the recording itself: 
* I see that with the trajectory taken, there are some missed loop closures that could have been useful to detect to remove more VIO drift from the map. I don't recommend Data Recorder mode unless you know already what are the best point of views to detect loop closures and force scanning that way. For large areas, it is possible to just reduce the number of points/polygons shown while doing a normal mapping session to still keep online confirmation that loop closures can be detected. That removes the issue when we blindly record data and we try to fix loop closures afterwards offline, we see that if the camera has rotated slightly more at some point when revisiting close-by area already mapped, that would have help to detect a loop closure and remove the double wall effect.
* Setting depth confidence to low in iOS app settings could help to find loop closures where the LiDAR cannot see any depth.
* I tried to fix some drift by reprocessing with SIFT features, however there are some double walls I could not remove because the point of views were too different between the two traversals:
   ```
   rtabmap-reprocess -default \
       --Mem/UseOdomFeatures false \
       --Kp/DetectorStrategy 1 \
       --Vis/FeatureType 1 \
       --RGBD/OptimizeMaxError 8 \
       --Vis/MinInliers 15 \
       --Mem/DepthAsMask false \
       --Vis/CorNNDR 0.6 \
       --Kp/NndrRatio 0.6 \
       rtabmap_db_reprocess_issue.db output.db
   ```

hkortier, 2025-03-18:
Thanks for your detailed explanation and advice @matlabbe! 

I'm cheazers from the rtabmap forum hence the similarities in questions.. :)

Would, besides the type of feature detector/descriptor, changing the following settings help in detecting more loop closures:
- Increase the WM size (how to do this?)
- Increase the STM size
- Altering the Bayesian hypothesis mechanism, e.g. lowering detection thresholds or transition probabilities. 

Or is is more likely that false-positives will dominate then?

In our use case is offline processing totally fine.

matlabbe, 2025-03-19:
With `Rtabmap/MemoryThr=0`, the WM size is unlimited, meaning you cannot miss a loop closure because it was in long-term memory.

Increasing the STM size only helps if you want to avoid detecting loop closures with images just seen, which could skew high probabilities on these images, hiding the real loop closures on older images that we really want to detect. Based on https://arxiv.org/abs/2407.15304:

> STM is used to observe similarities through time between
> consecutive images for weight updates, while the role of the
> WM is to detect loop closures between locations in space.
> Similar to [17], RTAB-Map does not use locations in STM
> to avoid loop closures on locations that have just been visited
> (because most of the time, the last location frequently looks
> similar to the most recent ones). The STM size is
> set based on the robot velocity and the rate at which the
> locations are acquired

Changing the transition probabilities may be difficult to tune, so I would recommend to tune `Rtabmap/LoopThr` instead. Lowering this threshold can generate more loop closure detections. Beware that while Recall can increase, the Precision can be lowered (more false positives). However, with all mechanisms RTAB-Map has to reject bad loop closures, you could reduce it from its default (0.11).

Another parameter to accept more loop closures is to decrease `Vis/MinInliers` (default 20) parameter, but again, at the risk of decreasing Precision.

Note that in my previous above, when using float visual descriptors, `Kp/NndrRatio` should be `0.6`, while binary descriptors it should be `0.8`.

You can also experiment with `Kp/TfIdfLikelihoodUsed` set to `false` (default `true`), which does a direct comparison between signatures instead of using inverted file index, though is is slower.

hkortier, 2025-03-19:
Thank you so much for the detailed advice!

closed 2025-03-19 by hkortier