introlab/rtabmap #1471: curl does not follow redirects correctly

hkortier, 2025-03-18:
https://github.com/introlab/rtabmap/blob/5f946cc26b610be8dc66504ed5e557d92c948fbe/app/ios/RTABMapApp/install_deps.sh#L34

Advice is to use wget: 

`wget -O boost_1_59_0.tar.gz https://downloads.sourceforge.net/project/boost/boost/1.59.0/boost_1_59_0.tar.gz`

matlabbe, 2025-03-19:
I just did it on linux (curl 7.81.0) and it downloads correctly. What version of curl is installed on your computer?

hkortier, 2025-03-19:
```
curl -V
curl 8.7.1 (x86_64-apple-darwin24.0) libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.63.0
Release-Date: 2024-03-27
```

matlabbe, 2025-03-20:
I tried tag 8.7.1 from source and it looked okay too
```
root@b433ac2ab4a7:/curl/build/src# ./curl -V
curl 8.7.0-DEV (Linux) libcurl/8.7.0-DEV OpenSSL/3.0.13 brotli/1.1.0 zstd/1.5.5 libpsl/0.21.2 nghttp2/1.59.0
Release-Date: [unreleased]
Protocols: dict file ftp ftps gopher gophers http https imap imaps ipfs ipns mqtt pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS brotli HSTS HTTP2 HTTPS-proxy IPv6 Largefile NTLM PSL SSL threadsafe TLS-SRP UnixSockets zstd
root@b433ac2ab4a7:/curl/build/src# ./curl -L https://downloads.sourceforge.net/project/boost/boost/1.59.0/boost_1_59_0.tar.gz -o boost_1_59_0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   371  100   371    0     0   2740      0 --:--:-- --:--:-- --:--:--  2748
100 79.8M  100 79.8M    0     0  17.4M      0  0:00:04  0:00:04 --:--:-- 18.7M
root@b433ac2ab4a7:/curl/build/src# ls -lh boost_1_59_0.tar.gz 
-rw-r--r-- 1 root root 80M Mar 20 03:37 boost_1_59_0.tar.gz
```

I'll still tag this issue to update to `wget` and test on my mac.