introlab/rtabmap #1473: rtabmap build on windows

xxxy12, 2025-03-25:
When I build rtabmap from source on windows,  an error happend: External symbols that cannot be resolved "protected: static class rtabmap::RtabmapColorOcTree::StaticMemberInitializer rtabmap::RtabmapColorOcTree::RtabmapColorOcTreeMemberInit" (?RtabmapColorOcTreeMemberInit@RtabmapColorOcTree@rtabmap@@1VStaticMemberInitializer@12@A). So I commented out lines 285-288 of OctoMap.cpp, and directly defines the RtabmapColorOcTree: : StaticMemberInitializer RtabmapColorOcTree: : RtabmapColorOcTreeMemberInit in 283 , as shown in the figure below.

[image]
But when I finished build and Install, I ran rtabmap.exe and it didn't work properly. I find a instruction  where I changed: On Windows, the app freezes on start if the following is defined. Does the problem with rtabmap realted to the change. If so, how can I sove the error External symbols that cannot be resolved...

[image]

xxxy12, 2025-03-26:
Here is my cmake log:
Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19045.
MOBILE_BUILD=OFF
CMake Warning (dev) at D:/software/PCL/cmake/PCLConfig.cmake:145 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable EIGEN_ROOT is set to:

    D:/software/PCL_dep/Eigen3/include/eigen3

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  D:/software/PCL/cmake/PCLConfig.cmake:298 (find_eigen)
  D:/software/PCL/cmake/PCLConfig.cmake:544 (find_external_library)
  CMakeLists.txt:234 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Eigen found (include: D:/software/PCL_dep/Eigen3/include/eigen3, version: 3.4.0)
CMake Warning (dev) at D:/software/PCL/cmake/PCLConfig.cmake:127 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  D:/software/PCL/cmake/PCLConfig.cmake:296 (find_boost)
  D:/software/PCL/cmake/PCLConfig.cmake:544 (find_external_library)
  CMakeLists.txt:234 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/software/PCL/cmake/PCLConfig.cmake:127 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  D:/software/PCL/cmake/PCLConfig.cmake:296 (find_boost)
  D:/software/PCL/cmake/PCLConfig.cmake:544 (find_external_library)
  CMakeLists.txt:234 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Eigen found (include: D:/software/PCL_dep/Eigen3/include/eigen3, version: 3.4.0)
CMake Warning (dev) at D:/software/PCL/cmake/Modules/FindFLANN.cmake:44 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable FLANN_ROOT is set to:

    D:/software/PCL_dep/flann

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  D:/software/PCL/cmake/PCLConfig.cmake:255 (find_package)
  D:/software/PCL/cmake/PCLConfig.cmake:300 (find_flann)
  D:/software/PCL/cmake/PCLConfig.cmake:544 (find_external_library)
  CMakeLists.txt:234 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Found flann version 1.9.2
Found Qhull version 8.0.2
PCL compile options don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
Found Sqlite3: D:/software/vcpkg/installed/x64-windows/include D:/software/vcpkg/installed/x64-windows/lib/sqlite3.lib
Found OpenMP: 
Add PCL_OMP to definitions
Found OpenCV: D:/software/OPENCV/include
Found PCL: D:/software/PCL/include/pcl-1.13;D:/software/PCL_dep/Eigen3/include/eigen3;D:/software/PCL_dep/boost/include/boost-1_78
[... 335 lines cut ...]

  The target yaml-cpp is deprecated and will be removed in version 0.10.0.
  Use the yaml-cpp::yaml-cpp target instead.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at tools/EurocDataset/CMakeLists.txt:31 (TARGET_LINK_LIBRARIES):
  The library that is being linked to, yaml-cpp, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  The target yaml-cpp is deprecated and will be removed in version 0.10.0.
  Use the yaml-cpp::yaml-cpp target instead.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at tools/EurocDataset/CMakeLists.txt:31 (TARGET_LINK_LIBRARIES):
  The library that is being linked to, yaml-cpp, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  The target yaml-cpp is deprecated and will be removed in version 0.10.0.
  Use the yaml-cpp::yaml-cpp target instead.

This warning is for project developers.  Use -Wno-dev to suppress it.

Generating done (2.2s)

matlabbe, 2025-03-31:
If you don't need OctoMap, don't build with it:
```
cmake -DWITH_OCTOMAP=OFF ..
```
If you need it, you could try a newer version of RTAB-Map, it seems you are building 0.21.0.

xxxy12, 2025-03-31:
Whether I build 0.21.0 or 0.21.4, the build succeeds only when WITH_OCTOMAP is set to OFF. However, when run RTABMap.exe, , I always face the problem that : Application cannot start properly(0xc0000142). I have added the bin directory of dependent environments such as OpenCV, PCL, and Qt to the system PATH of environment variable.

matlabbe, 2025-04-06:
Did you try building following instructions from https://github.com/introlab/rtabmap/wiki/Installation#vcpkg? 

Otherwise, I would need detailed instructions of how you built all dependencies.

xxxy12, 2025-04-07:
I built  all thedependencies from source code but got Application cannot start properly(0xc0000142). So I  built following instructions from https://github.com/introlab/rtabmap/wiki/Installation#vcpkg .However, when I finished compiling and ran the RTABMAP.exe, an error occurred  : Unable to locate program input point FT_Get_Color_Glyph_ClipBox on dynamic link library D:\software\vcpkg\installed\x64-windows\bin\harfbuzz.dll

xxxy12, 2025-04-10:
Here is the CMake log with all  deps from vcpkg:
```
Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
MOBILE_BUILD=OFF
Found OpenMP: TRUE (found version "2.0") found components: CXX
CMake Warning (dev) at D:/software/vcpkg/installed/x64-windows/share/pcl/Modules/FindFLANN.cmake:45 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable FLANN_ROOT is set to:

    D:/software/vcpkg/installed/x64-windows

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  D:/software/vcpkg/installed/x64-windows/share/pcl/PCLConfig.cmake:242 (find_package)
  D:/software/vcpkg/installed/x64-windows/share/pcl/PCLConfig.cmake:287 (find_flann)
  D:/software/vcpkg/installed/x64-windows/share/pcl/PCLConfig.cmake:563 (find_external_library)
  CMakeLists.txt:238 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Found flann version 1.9.2
Found Qhull version 8.0.2
PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
Found Sqlite3: D:/software/vcpkg/installed/x64-windows/include D:/software/vcpkg/installed/x64-windows/lib/sqlite3.lib
Found OpenMP: TRUE (found version "2.0") found components: C CXX
Found OpenMP: 
Add PCL_OMP to definitions
Found OpenCV: D:/software/vcpkg/installed/x64-windows/include/opencv4
Found PCL: D:/software/vcpkg/installed/x64-windows/include;D:/software/vcpkg/installed/x64-windows/include/eigen3
Found ZLIB: D:/software/vcpkg/installed/x64-windows/include
Searching for PEGTL
Searching for PEGTL - found target taocpp::pegtl
CMake Warning (dev) at D:/Program Files/CMake/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:441 (message):
  The package name passed to `find_package_handle_standard_args` (NetCDF)
  does not match the name of the calling package (netCDF).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  D:/software/vcpkg/installed/x64-windows/share/vtk/FindNetCDF.cmake:33 (find_package_handle_standard_args)
  D:/Program Files/CMake/share/cmake-3.30/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  D:/software/vcpkg/installed/x64-windows/share/external_packages/TPL-Seacas-Netcdf/TPL-Seacas-NetcdfConfig.cmake:4 (find_dependency)
  D:/software/vcpkg/installed/x64-windows/share/cmake/SEACASExodus/SEACASExodusConfig.cmake:163 (include)
  D:/software/vcpkg/installed/x64-windows/share/cmake/SEACASIoss/SEACASIossConfig.cmake:174 (include)
  D:/software/vcpkg/installed/x64-windows/share/vtk/VTK-vtk-module-find-packages.cmake:685 (find_package)
  D:/software/vcpkg/installed/x64-windows/share/vtk/vtk-config.cmake:162 (include)
  CMakeLists.txt:298 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Found Sqlite3: D:/software/vcpkg/installed/x64-windows/include D:/software/vcpkg/installed/x64-windows/lib/sqlite3.lib
Could not find the VTK package due to a missing dependency: SQLite3
VTK>=9 detected, will use VTK_QT_VERSION=6 for Qt version.
VTK_RENDERING_BACKEND=
CMake Warning at cmake_modules/FindKinectSDK2.cmake:71 (message):
  Directory "" not found.
Call Stack (most recent call first):
  cmake_modules/FindKinectSDK2.cmake:119 (check_dir)
  CMakeLists.txt:444 (FIND_PACKAGE)

KinectSDK2_FOUND : FALSE
FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
--------------------------------------------
Info :
  RTAB-Map Version =     0.21.4
[... 51 lines cut ...]
  With dc1394               = NO (dc1394 not found)
  With FlyCapture2/Triclops = NO (Point Grey SDK not found)
  With ZED                  = NO (ZED sdk and/or cuda not found)
  With ZEDOC                = NO (ZED Open Capture not found)
  With RealSense            = NO (librealsense not found)
  With RealSense2           = NO (librealsense2 not found)
  With MyntEyeS             = NO (mynteye s sdk not found)
  With DepthAI              = NO (WITH_DEPTHAI=OFF)

 Odometry Approaches:
  With loam_velodyne        = NO (WITH_LOAM=OFF)
  With floam                = NO (WITH_FLOAM=OFF)
  With libfovis             = NO (WITH_FOVIS=OFF)
  With libviso2             = NO (WITH_VISO2=OFF)
  With dvo_core             = NO (WITH_DVO=OFF)
  With okvis                = NO (WITH_OKVIS=OFF)
  With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
  With VINS-Fusion          = NO (WITH_VINS=OFF)
  With OpenVINS             = NO (WITH_OPENVINS=OFF)
  With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
Show all options with: cmake -LA | grep WITH_
--------------------------------------------
Configuring done (4.2s)
Generating done (5.2s)
```

matlabbe, 2025-04-12:
I tried with latest vcpkg version and it builds correctly. Here the cmake I have:
```
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22631.
-- MOBILE_BUILD=OFF
-- Found OpenMP: TRUE (found version "2.0") found components: CXX
-- Found flann version 1.9.2
-- Found Qhull version 8.0.2
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found SQLite3: C:/vcpkg/installed/x64-windows/include C:/vcpkg/installed/x64-windows/lib/sqlite3.lib
-- Found OpenMP: TRUE (found version "2.0") found components: C CXX
-- Found OpenMP:
-- Add PCL_OMP to definitions
-- Found OpenCV: C:/workspace/opencv/build/install/include
-- Found PCL: C:/vcpkg/installed/x64-windows/include;C:/vcpkg/installed/x64-windows/include/eigen3
-- Found ZLIB: C:/vcpkg/installed/x64-windows/include
-- Searching for PEGTL
-- Searching for PEGTL - found target taocpp::pegtl
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (NetCDF)
  does not match the name of the calling package (netCDF).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/vcpkg/installed/x64-windows/share/vtk/FindNetCDF.cmake:33 (find_package_handle_standard_args)
  C:/Program Files/CMake/share/cmake-3.24/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  C:/vcpkg/installed/x64-windows/share/external_packages/TPL-Seacas-Netcdf/TPL-Seacas-NetcdfConfig.cmake:4 (find_dependency)
  C:/vcpkg/installed/x64-windows/share/cmake/SEACASExodus/SEACASExodusConfig.cmake:163 (include)
  C:/vcpkg/installed/x64-windows/share/cmake/SEACASIoss/SEACASIossConfig.cmake:174 (include)
  C:/vcpkg/installed/x64-windows/share/vtk/VTK-vtk-module-find-packages.cmake:685 (find_package)
  C:/vcpkg/installed/x64-windows/share/vtk/vtk-config.cmake:162 (include)
  CMakeLists.txt:300 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (NetCDF)
  does not match the name of the calling package (netCDF).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/vcpkg/installed/x64-windows/share/vtk/FindNetCDF.cmake:33 (find_package_handle_standard_args)
  C:/Program Files/CMake/share/cmake-3.24/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  C:/vcpkg/installed/x64-windows/share/external_packages/TPL-Seacas-Netcdf/TPL-Seacas-NetcdfConfig.cmake:4 (find_dependency)
  C:/vcpkg/installed/x64-windows/share/cmake/SEACASExodus/SEACASExodusConfig.cmake:163 (include)
  C:/vcpkg/installed/x64-windows/share/vtk/VTK-vtk-module-find-packages.cmake:779 (find_package)
  C:/vcpkg/installed/x64-windows/share/vtk/vtk-config.cmake:162 (include)
  CMakeLists.txt:300 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: opengl32  found components: OpenGL
-- VTK>=9 detected, will use VTK_QT_VERSION=6 for Qt version.
-- VTK_RENDERING_BACKEND=
-- Found freenect2: C:/workspace/libfreenect2-0.2.1/build/install/include
-- KinectSDK2_FOUND : TRUE
-- Found Kinect for Windows 2: C:/Program Files/Microsoft SDKs/Kinect/v2.0_1409/inc
-- Found Kinect for Azure: C:/Program Files/Azure Kinect SDK v1.4.1/sdk/include
-- Found OpenNI2: C:/Program Files/OpenNI2/Include
-- Found OpenGL: opengl32
-- Found g2o (targets)
-- g2o include dir: C:/vcpkg/installed/x64-windows/include/g2o/core;C:/vcpkg/installed/x64-windows/include
-- Latest g2o factory version detected with shared ptr (factory file: C:/vcpkg/installed/x64-windows/include/g2o/core/factory.h).
-- GTSAM include directory:  C:/Libraries/CMake/../include;C:/vcpkg/installed/x64-windows/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found Boost: C:/vcpkg/installed/x64-windows/share/boost/BoostConfig.cmake (found version "1.87.0") found components: thread filesystem system program_options date_time
-- Found Boost: C:/vcpkg/installed/x64-windows/share/boost/BoostConfig.cmake (found version "1.87.0") found components: thread filesystem system program_options date_time chrono
-- Found libpointmatcher: C:/Libraries/include;C:/vcpkg/installed/x64-windows/include/eigen3
-- Found Boost: C:/vcpkg/installed/x64-windows/share/boost/BoostConfig.cmake (found version "1.87.0") found components: thread filesystem system program_options date_time
[... 62 lines cut ...]
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
--   With XVisio SDK           = NO (WITH_XVSDK=OFF)
--
[... cut ...]
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/workspace/rtabmap/build
```
Note that I am using a custom opencv version, but that shouldn't matter. Also, if you don't have this commit https://github.com/introlab/rtabmap/commit/08f031e11c45589fc2b68440383a3e40982dc06f, it won't build on latest vcpkg because of the new boost version 1.87. I assume that you don't have latest vcpkg version.

xxxy12, 2025-05-12:
The bug  Application cannot start properly(0xc0000142) only appears in the exe with gui and was found to be related to QT .The incorrect QT dynamic library was found when running exe.

closed 2025-05-12 by xxxy12