introlab/rtabmap #1478: D405 crashes program

pepper11com, 2025-04-01:
I have installed the RealSense2 2.54.2 SDK and the latest version of RTABMap but when I connect my Realsense D405 and click on the "play" button the program crashes with no warning. I checked the Event Viewer and it gives me an error. I already reinstalled all the latest Microsoft Visual C++ Redistributable Versions so im not sure why this is happening, any tips? 

-	System
-	Provider
[ Name]	Application Error
[ Guid]	{a0e9b465-b939-57d7-b27d-95d8e925ff57}
EventID	1000
Version	0
Level	2
Task	100
Opcode	0
Keywords	0x8000000000000000
-	TimeCreated
[ SystemTime]	2025-04-01T02:11:56.2861402Z
EventRecordID	81615
Correlation
-	Execution
[ ProcessID]	26144
[ ThreadID]	9272
Channel	Application
Computer	LAPTOP-xxx
-	Security
[ UserID]	xxx
-	EventData
AppName	RTABMap.exe
AppVersion	0.0.0.0
AppTimeStamp	64180fdd
ModuleName	ucrtbase.dll
ModuleVersion	10.0.22621.3640
ModuleTimeStamp	10c46e71
ExceptionCode	c0000409
FaultingOffset	000000000007f6fe
ProcessId	0x2320
ProcessCreationTime	0x1dba2ab50861dd4
AppPath	C:\Program Files\RTABMap\bin\RTABMap.exe
ModulePath	C:\WINDOWS\System32\ucrtbase.dll
IntegratorReportId	22f9d633-af87-4b3f-a345-ea15216e1e45
PackageFullName	
PackageRelativeAppId

matlabbe, 2025-04-06:
I cannot reproduce the issue with this version: https://github.com/introlab/rtabmap/releases/tag/0.21.4 (with a D435i camera updated with latest firmware)

"the latest version of RTABMap"... do you mean latest release?

pepper11com, 2025-04-08:
> I cannot reproduce the issue with this version: https://github.com/introlab/rtabmap/releases/tag/0.21.4 (with a D435i camera updated with latest firmware)
> 
> "the latest version of RTABMap"... do you mean latest release?

I see the issue is its looking for a emitter even though the D405 doesnt have one, amd when I turn all the optional boxes off in Source it still gives me the emitter error...

pepper11com, 2025-04-08:
> I cannot reproduce the issue with this version: https://github.com/introlab/rtabmap/releases/tag/0.21.4 (with a D435i camera updated with latest firmware)
> 
> "the latest version of RTABMap"... do you mean latest release?

matlabbe, 2025-04-09:
Did you see my comment here? Maybe worth the try if you build from source: https://github.com/introlab/rtabmap/issues/1480#issuecomment-2781710279

I wasn't aware there were realsense cameras without emitter. I'll need to check if we can know if he device supports it from the API before trying to assign a value to it, or catch the error and silently ignore it if possible.

commit 43c398a0 (2025-04-14): Fixed #1478
closed 2025-04-14 by matlabbe