introlab/rtabmap #1484: Outdoor mapping problem

zhangle0803, 2025-04-14:
Recently, when I was using realsensed415 for outdoor mapping, the rtabmap algorithm was affected by the lighting, resulting in poor mapping effect. There were many black spots (obstacles) on the raster map that did not exist originally. How should I solve this problem? Thank you

[image]
[image]
[image]

zhangle0803, 2025-04-14:
The first picture is the construction under cloudy conditions, and the second picture is the result of the construction under sunny conditions. It can be seen that there are many obstacles that should not exist in the middle of the second picture. How should I solve this problem? Thank you all

matlabbe, 2025-04-19:
I assume you are using depth computed from the camera. There could be settings to avoid bad disparity you are seeing. See https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

On rtabmap side, the depth noise can be filtered with `Grid/` parameters:
```
rtabmap --params | grep Grid/
Param: Grid/3D = "true"                                    [A 3D occupancy grid is required if you want an OctoMap (3D ray tracing). Set to false if you want only a 2D map, the cloud will be projected on xy plane. A 2D map can be still generated if checked, but it requires more memory and time to generate it. Ignored if laser scan is 2D and "Grid/Sensor" is 0.]
Param: Grid/CellSize = "0.05"                              [Resolution of the occupancy grid.]
Param: Grid/ClusterRadius = "0.1"                          [[Grid/NormalsSegmentation=true] Cluster maximum radius.]
Param: Grid/DepthDecimation = "4"                          [[Grid/DepthDecimation=true] Decimation of the depth image before creating cloud.]
Param: Grid/DepthRoiRatios = "0.0 0.0 0.0 0.0"             [[Grid/Sensor>=1] Region of interest ratios [left, right, top, bottom].]
Param: Grid/FlatObstacleDetected = "true"                  [[Grid/NormalsSegmentation=true] Flat obstacles detected.]
Param: Grid/FootprintHeight = "0.0"                        [Footprint height used to filter points over the footprint of the robot. Footprint length and width should be set.]
Param: Grid/FootprintLength = "0.0"                        [Footprint length used to filter points over the footprint of the robot.]
Param: Grid/FootprintWidth = "0.0"                         [Footprint width used to filter points over the footprint of the robot. Footprint length should be set.]
Param: Grid/GroundIsObstacle = "false"                     [[Grid/3D=true] Ground segmentation (Grid/NormalsSegmentation) is ignored, all points are obstacles. Use this only if you want an OctoMap with ground identified as an obstacle (e.g., with an UAV).]
Param: Grid/MapFrameProjection = "false"                   [Projection in map frame. On a 3D terrain and a fixed local camera transform (the cloud is created relative to ground), you may want to disable this to do the projection in robot frame instead.]
Param: Grid/MaxGroundAngle = "45"                          [[Grid/NormalsSegmentation=true] Maximum angle (degrees) between point's normal to ground's normal to label it as ground. Points with higher angle difference are considered as obstacles.]
Param: Grid/MaxGroundHeight = "0.0"                        [Maximum ground height (0=disabled). Should be set if "Grid/NormalsSegmentation" is false.]
Param: Grid/MaxObstacleHeight = "0.0"                      [Maximum obstacles height (0=disabled).]
Param: Grid/MinClusterSize = "10"                          [[Grid/NormalsSegmentation=true] Minimum cluster size to project the points.]
Param: Grid/MinGroundHeight = "0.0"                        [Minimum ground height (0=disabled).]
Param: Grid/NoiseFilteringMinNeighbors = "5"               [Noise filtering minimum neighbors.]
Param: Grid/NoiseFilteringRadius = "0.0"                   [Noise filtering radius (0=disabled). Done after segmentation.]
Param: Grid/NormalK = "20"                                 [[Grid/NormalsSegmentation=true] K neighbors to compute normals.]
Param: Grid/NormalsSegmentation = "true"                   [Segment ground from obstacles using point normals, otherwise a fast passthrough is used.]
Param: Grid/PreVoxelFiltering = "true"                     [Input cloud is downsampled by voxel filter (voxel size is "Grid/CellSize") before doing segmentation of obstacles and ground.]
Param: Grid/RangeMax = "5.0"                               [Maximum range from sensor. 0=inf.]
Param: Grid/RangeMin = "0.0"                               [Minimum range from sensor.]
Param: Grid/RayTracing = "false"                           [Ray tracing is done for each occupied cell, filling unknown space between the sensor and occupied cells. If Grid/3D=true, RTAB-Map should be built with OctoMap support, otherwise 3D ray tracing is ignored.]
Param: Grid/Scan2dUnknownSpaceFilled = "false"             [Unknown space filled. Only used with 2D laser scans. Use Grid/RangeMax to set maximum range if laser scan max range is to set.]
Param: Grid/ScanDecimation = "1"                           [[Grid/Sensor=0 or 2] Decimation of the laser scan before creating cloud.]
Param: Grid/Sensor = "1"                                   [Create occupancy grid from selected sensor: 0=laser scan, 1=depth image(s) or 2=both laser scan and depth image(s).]
```
I suggest to use `rtabmap-databaseViewer`, change `Grid` settings under Core Parameters panel, then click on Edit->Regenerate local grids...

zhangle0803, 2025-04-21:
> I assume you are using depth computed from the camera. There could be settings to avoid bad disparity you are seeing. See [...]

I want to know where these parameters can be modified? Is it in some file within rtabmap-ros or in rtabmap-database-Viwer? Exactly where?

matlabbe, 2025-04-27:
For ROS1: see http://wiki.ros.org/rtabmap_slam#rtabmap

For ROS2: we can pass them as arguments to nodes like in ROS1. To set them as ROS parameters, you should pass them with values as **strings**, like in this example:
https://github.com/introlab/rtabmap_ros/blob/f5a3e487056e399b834971ef6a64126843666f7f/rtabmap_examples/launch/rgbdslam_datasets.launch.py#L53-L59

zhangle0803, 2025-04-28:
> For ROS1: see http://wiki.ros.org/rtabmap_slam#rtabmap
>
> For ROS2: we can pass them as arguments to nodes like in ROS1. To set them as ROS parameters, you should pass them with values as **strings**, [...]

Thanks, and I'd like to know how I can create a map where dynamic objects go, such as pedestrians and bicycles passing by while building a map.

matlabbe, 2025-04-30:
Enabling `Grid/RayTracing` parameter can help to clear dynamic obstacles added to the map. Note that this only affects the occupancy grid topics (`grid_map` or `octomap`), not the default point cloud shown in 3D map view of rtabmap_viz or by MapCloud plugin in rviz.

matlabbe, 2025-06-01:
Default is GFTT-BRIEF when building rtabmap with opencv's xFeatures2D module:
```
Param: Kp/DetectorStrategy = "6"                           [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE 10=ORB-OCTREE 11=SuperPoint 12=SURF/FREAK 13=GFTT/DAISY 14=SURF/DAISY 15=PyDetector]
Param: Vis/FeatureType = "6"                               [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE 10=ORB-OCTREE 11=SuperPoint 12=SURF/FREAK 13=GFTT/DAISY 14=SURF/DAISY 15=PyDetector]
```
Otherwise these parameters will be 8: GFTT-ORB. To use ORB-OCTREE, set to 10 both parameters. In a launch file in ros1 melodic, you can do:
```xml
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
  <param name="Kp/DetectorStrategy" value="10"/>
  <param name="Vis/FeatureType" value="10"/>
  ...
```
If you are not sure of your build, explicitly set the parameters like above and you would see warnings if the feature is not available.

zhangle0803, 2025-06-04:
> Default is GFTT-BRIEF when building rtabmap with opencv's xFeatures2D module:
> 
> ```
> Param: Kp/DetectorStrategy = "6"                           [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE 10=ORB-OCTREE 11=SuperPoint 12=SURF/FREAK 13=GFTT/DAISY 14=SURF/DAISY 15=PyDetector]
> Param: Vis/FeatureType = "6"                               [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE 10=ORB-OCTREE 11=SuperPoint 12=SURF/FREAK 13=GFTT/DAISY 14=SURF/DAISY 15=PyDetector]
> ```
> 
> Otherwise these parameters will be 8: GFTT-ORB. To use ORB-OCTREE, set to 10 both parameters. In a launch file in ros1 melodic, you can do:
> 
> <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
>   <param name="Kp/DetectorStrategy" value="10"/>
>   <param name="Vis/FeatureType" value="10"/>
>   ...
> 
> If you are not sure of your build, explicitly set the parameters like above and you would see warnings if the feature is not available.

<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
  <param name="Kp/DetectorStrategy" value="10"/>
  <param name="Vis/FeatureType" value="10"/>
  I'd like to know in which specific file these codes should be modified? Thank you for your answer.

closed 2025-06-04 by zhangle0803

matlabbe, 2025-06-07:
You should not have to change any source file to use a different detector, if you use rosparam like above.

If you want to add a new detector, then you would have to modify:
https://github.com/introlab/rtabmap/blob/80d463ed20d78b1a5b3d0f7c24f19345696ab139/corelib/src/Features2d.cpp#L614

to return your own feature detector.

zhangle0803, 2025-06-23:
> You should not have to change any source file to use a different detector, if you use rosparam like above.
> 
> If you want to add a new detector, then you would have to modify:
> 
> [rtabmap/corelib/src/Features2d.cpp](https://github.com/introlab/rtabmap/blob/80d463ed20d78b1a5b3d0f7c24f19345696ab139/corelib/src/Features2d.cpp#L614)
> 
> Line 614 in [80d463e](/introlab/rtabmap/commit/80d463ed20d78b1a5b3d0f7c24f19345696ab139)
>  Feature2D * Feature2D::create(Feature2D::Type type, const ParametersMap & parameters) 
> 
> to return your own feature detector.

I want to know if you only need to add new features detection algorithm in rtabmap/corelib/SRC/Features2d CPP modify the code in this file? Or modify more files? If possible, could you give an example and provide a specific modification method?thank you very much！

matlabbe, 2025-06-29:
For a minimal working example so that you can use `Kp/DetectorStrategy` and `Vis/FeatureType` for your approach. You would need to create a new class (inheriting `Feature2D`) under `Features2D.cpp` and `Features2D.h` (with new type in the `Feature2D::Type` enum). 

Optional: 
* You may update the description of these two parameters in `Parameters.h` to tell which index is your approach.
* Note that if you need to include a new dependency, you would have to update the main `CMakeLists.txt` and the one under `corelib/src`.
* If you care about rtabmap's ui, you would need to update `preferencesDialog.ui` (with Qt Designer) and adjust the `QComboBox` values of the two parameters.

zhangle0803, 2025-07-10:
> For a minimal working example so that you can use `Kp/DetectorStrategy` and `Vis/FeatureType` for your approach. You would need to create a new class (inheriting `Feature2D`) under `Features2D.cpp` [...]

[image]

this is my cmake/ result.please help me!!!

zhangle0803, 2025-07-21:
> There could be some compilation issues with older EOL ubuntu versions. I'll suggest to use docker. Here an example of building rtabmap_ros (ros1) with SuperPoint/SuperGlue support: https://github.com/introlab/rtabmap_ros/blob/master/docker/noetic/superpoint/Dockerfile

Hello, my system is Melodic. Recently, I have been using ORB-SLAM3 as the external odometry for RTAB-Map. I have successfully built RTAB-Map with ORB-SLAM3 v0.4, as well as rtabmap_ros, orbslam_ros. But now, how can I combine it with the RealSense D455? What launch file should I run? Or do I need to write one myself? If I need to write one myself, is there any reference? Please help me. Thank you!

zhangle0803, 2025-07-21:
> There could be some compilation issues with older EOL ubuntu versions. I'll suggest to use docker. Here an example of building rtabmap_ros (ros1) with SuperPoint/SuperGlue support: https://github.com/introlab/rtabmap_ros/blob/master/docker/noetic/superpoint/Dockerfile

There's another issue! After I ran the commands `roslaunch rtabmap_ros rtabmap.launch args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/king/rtabmap_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt" MH_seq:=true raw_images_for_odom:=true` and `roslaunch realsense2_camera rs_camera.launch`, an error occurred:

[rtabmap/rgbd_odometry-2] process has died [pid 14407, exit code -11, cmd /home/king/rtabmap_ws/devel/lib/rtabmap_ros/rgbd_odometry Odom/Strategy 5 OdomORBSLAM/VocPath /home/king/rtabmap_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay odom:=/rtabmap/odom imu:=/imu/data __name:=rgbd_odometry __log:=/home/king/.ros/log/e9fc04c0-662e-11f0-b18c-00430500bcc7/rtabmap-rgbd_odometry-2.log].
log file: /home/king/.ros/log/e9fc04c0-662e-11f0-b18c-00430500bcc7/rtabmap-rgbd_odometry-2*.log
[ WARN] [1753101229.277753509]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=20).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info
How can it be solved?

matlabbe, 2025-08-03:
Eigen issues like that are usually caused by using `-march=native` when building orbslam3, then included as library inside rtabmap. Remove all `-march=native` when building orbslam3.

zhangle0803, 2025-09-09:
> Eigen issues like that are usually caused by using `-march=native` when building orbslam3, then included as library inside rtabmap. Remove all `-march=native` when building orbslam3. The [gist I linked](https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13?permalink_comment_id=3763035#gistcomment-3763035) shows an example.
Hello, I recently combined orbslam3 as an external odometry with rtabmap to replace the built-in odom of rtab. I want to know what the theoretical support for this approach is? Is it the entire slam process of orbslam3 combined with rtabmap, or is it a certain part of the orbslam algorithm? Or was the orb descriptor used? I want to figure out the principle of this theory. Thank you!

matlabbe, 2025-09-10:
With ORB_SLAM3 inside rtabmap, we disabled ORB_SLAM3 loop closure management to just run the visual odometry. RTAB-Map manages then the map's graph and loop closure detection/optimization.

If you want to use ORB_SLAM3 externally, but feeding its odometry topic to rtabmap node, make sure loop closure detection in ORB_SLAM3 is disabled, meaning the odometry topic provided by ORB_SLAM3 will never jump.

lifafang, 2025-11-03:
How should I run an external ORB-SLAM3 instance and achieve full IMU support by subscribing to ORB-SLAM3's odometer topic? Could you provide a case study or a simple workflow overview? @matlabbe

zhangle0803, 2026-01-07:
> Which ORB-SLAM3 ROS2 implementation are you using? Unfortunately, the "most popular" ROS2 implementations I've found (https://github.com/zang09/ORB_SLAM3_ROS2 and https://github.com/Mechazo11/ros2_orb_slam3) don't publish any topic back on ROS2. At minimum, the package should publish odometry TF (e.g., `odom->base_link`) to be integrated externally with `rtabmap_ros`.
> 
> It seems the easiest way to integrate ORB_SLAM3 odometry under ROS2 is to [build rtabmap with ORB_SLAM3 support](https://github.com/introlab/rtabmap/pull/1152#issue-1967455635), then start `rtabmap_odom/rgbd_odometry` or `rtabmap_odom/stereo_odometry` node with these [parameters](https://github.com/introlab/rtabmap/issues/1104#issuecomment-1667997429):
> 
> "Odom/Strategy": "5"
> "OdomORBSLAM/VocPath": "PATH/TO/ORB_SLAM/Vocabulary/ORBvoc.txt"

When I was running rtabmap, originally using pure visual mapping, the camera was facing upwards towards the screen. But after adding the IMU, I don't know why the mapping direction shifted to the right.

matlabbe, 2026-01-12:
If the camera was facing upwards, then there is a missing TF optical rotation `-PI/2 0 -PI/2` between the camera base frame and the optical frame.