introlab/rtabmap #1485: [Bug] DBDriverSqlite3 does not save and load GlobalDescriptor correctly

borongyuan, 2025-04-14:
When the global descriptor is used for likelihood evaluation, I found that the system works well at the beginning, but fails to perform loop closure detection after running for a long time. At this time, the posterior probability of new place will always be very high, rejecting all loop closures. Check the raw likelihood and you will find that the similarity with some nodes is 0. This is clearly unusual for using global descriptors, and suggests that this is an issue releated to memory management.
[image]
So I checked the node data and found that their globalDescriptors were of incorrect types and therefore failed the check in the following code.
https://github.com/introlab/rtabmap/blob/b750e94eaa0c934c20d82b1e4a4a05a4b5ef23a1/corelib/src/Signature.cpp#L259
Since VLAD descriptor is used, the type number of these descriptors should be 1, but now their type number continues to grow. Further inspection found the problem was in DBDriverSqlite3.
https://github.com/introlab/rtabmap/blob/b750e94eaa0c934c20d82b1e4a4a05a4b5ef23a1/corelib/src/DBDriverSqlite3.cpp#L3562-L3577
https://github.com/introlab/rtabmap/blob/b750e94eaa0c934c20d82b1e4a4a05a4b5ef23a1/corelib/src/DBDriverSqlite3.cpp#L6788-L6790
It seems that the type of GlobalDescriptor is not saved to the database?

matlabbe, 2025-04-14:
Indeed, the type was not saved. I fixed it in the commit above. I don't have a way to test this right now, so if you give a try and it works, you can close this issue.

matlabbe, 2025-04-19:
There could be optimization if we knew that some global descriptors have always the same size and the same number in the signature. I originally added this option for this [work](https://github.com/MarvinStuede/cmr_lidarloop), where if I remember well, the descriptors were not always the same size. From database point of view, by combining all descriptors together in same field, we could maybe save space / time by compressing/decompressing them all together, though updates would be more tedious than currently just adding a new row to a table.

For the WM and LTM split, the original idea about memory management is that we should never compare descriptors from LTM. When data is in LTM, it cannot be used for any online updates. Another way to see the problem is to disable memory management, so that all data is always in WM, so in this case, we could want to change how the global descriptors are stored in WM to make it faster if we want to support larger maps online. One similar optimization example I did many years ago (cannot find the commit it was prior to github repo) was the Bayes Filter prior update matrix, where instead of managing many sparse matrices, it was more efficient to build a big dense matrix and multiply once (even if most values are zeros in it). I can see with global descriptors that a similar approach could be to hold all global descriptors in a big matrix in WM to make the evaluation of similarity faster.

borongyuan, 2025-04-21:
> For the WM and LTM split, the original idea about memory management is that we should never compare descriptors from LTM.

This is actually where I have some thoughts. When I first used VLAD for similarity evaluation, I thought it could play a role similar to GPS. Because in NetVLAD training, GPS also provides weak supervision information. Both GPS and VLAD descriptors can help us quickly find the ones that may cause loop closure among a large number of nodes without the need for memory management. If memory management is enabled and the nodes found are in the LTM, they can be retrieved before Bayes filtering. But after checking I found that GPS is only used for likelihood filtering and graph optimization. Retrieval is only performed after Bayes filtering. A node in the LTM may be retrieved only if it has neighbors remaining in the WM. For a well-tuned system, this can be satisfied most of the time. You also mentioned in your paper that the retrieval step is time consuming. Therefore I am not sure whether it is worth adding the option to perform retrieval before Bayes filtering. If it is not for this feature, then there is not much need to store global descriptors separately.

> One similar optimization example I did many years ago (cannot find the commit it was prior to github repo) was the Bayes Filter prior update matrix,

Are you referring to Bayes/FullPredictionUpdate? I saw this enabled in some early examples, but it's now set to false by default, as well as your reply in #623, so I haven't tried this yet.

matlabbe, 2025-04-27:
For the Bayes filter, that was related to the prediction update, where I tried to use OpenCV `cv::SparseMat` instead of standard OpenCV `cv::Mat`, to try similar optimization from [FAB-Map 2.0](https://www.roboticsproceedings.org/rss05/p39.pdf). I think it was related to how good the optimization is on `cv::Mat` multiplications versus `cv::SparseMat` multiplication.

Note that `Bayes/FullPredictionUpdate` means reconstructing the whole prediction matrix of nodes contained only in WM.

What is costly when doing retrieval (LTM -> WM) is to rematch visual words to current state of the vocabulary and maybe re-balance the vocabulary. When using a fixed vocabulary, it is a lot faster thought, no re-matching needed. If we use occupancy grid, we still need to re-update the occupancy grid in both cases.

For the GPS, we didn't integrate it to memory management approach, but it could be interesting to do so to retrieve nodes from LTM close by current GPS position even if their neighbor nodes are not in WM. I remember someone asking me some years ago if it could be useful to integrate [spatialite](https://www.gaia-gis.it/fossil/libspatialite/index) to rtabmap, which is a tool to query efficiently data from a database that has spatial data. Back in the days, the GPS was not yet integrated to rtabmap, and I said "not really" because poses in LTM are not "global", but relative to nodes in WM. With GPS saved in database, that could be now a great way to use spatialite.

Assuming that we were already using GPS to retrieve from LTM all close nodes to current position (retrieved before Bayes update), to use global descriptors a similar way, we would probably need to keep all global descriptors in RAM. The idea is to compare current global descriptors with all previous locations to get a set of nodes with best similarity, then if they are in LTM, we retrieve them before doing Bayes update. While GPS could be probably implemented with spatialite to avoid keeping all GPS from all nodes in RAM, the issue with global descriptors is that matrix containing all descriptors would increase out of bounds in RAM. Even if it does, it doesn't mean that solution is not practically viable. In most practical uses, I don't even enable memory management of RTAB-Map because the map is not big enough, and I can get away with single mapping session, doing localization-only afterwards, then everything can stay in RAM (or WM).

cheers,
Mathieu

borongyuan, 2025-04-28:
Thanks for your detailed explanation. There appear to be several possible subsequent improvements involved. I'll close this for now. We can discuss the specific implementation in other issues or PRs later.

closed 2025-04-28 by borongyuan