introlab/rtabmap #1486: The result of iOS write ExportedMesh is inconsistent with reality

zhangyakai, 2025-04-18:
[image]

[image]

Why is the direction inconsistent when saving as an OBJ file using the write ExportedMesh Native function?

matlabbe, 2025-04-20:
Is the first view the OBJ shown in another 3D visualization app? I think the difference is that the rendering in the App is done in OpenGL world coordinates, while when we export, the map is converted in [ROS world coordinates (or ENU world coordinates)](https://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions).

zhangyakai, 2025-04-22:
I used the RTABMAP sample program for iOS to export an obj format model The same coordinate flipping problem occurs when using the built-in 3D viewing on Mac
I did not find any code related to coordinate settings in the API
How can I configure it to make the exported obj model consistent with the model rendered by OpenGL?
Is it still impossible to modify in the code, and can only be modified in the model tool software? @matlabbe

matlabbe, 2025-04-27:
We could add an option to export in a different coordinate system (I added a feature request label about it), though it is kinda a low priority feature as you can transform it in any coordinate system with CloudCompare or MeshLab.

Looking at the code, it seems I had the same issue when exporting to [Sketchfab](https://sketchfab.com/matlabbe):
https://github.com/introlab/rtabmap/blob/def99328ce22dbf5b0524b734dd946d022cf7a1e/app/android/jni/RTABMapApp.cpp#L4278-L4282

I think I commented it back because there is an option in Sketchfab to rotate the model.