introlab/rtabmap #1487: 2D (Diffdrive) Robot and elevators

sandeepdutta, 2025-04-18:
Hi Mathieu,

I am using RtabMap on a mobile platform with two Depth Cameras (one facing forward, the other facing backwards).
One use case for the robot is to get into an elevator and get off on a different floor, the robot has no way of sensing the
elevator ride, i.e. the odometry does not change during the ride.
Do you have any suggestions of how to handle mapping and localization, having multiple maps is possible. Is it possible
to handle the situation automatically ?

Thank you
Sandeep

matlabbe, 2025-04-20:
You may read this post: http://official-rtab-map-forum.206.s1.nabble.com/Rtab-map-in-different-floors-td6947.html

There are some pros and cons about using same database for all floors and split databases per floor.

sandeepdutta, 2025-04-28:
Hello Mathieu,

Thank you for the response and the link. I have ordered barometric pressure sensors. Will start  experiments soon.
Best Regards
Sandeep