introlab/rtabmap #1493: iOS Localization Mode Current Position

sb-msoe, 2025-04-28:
In the iOS app in localization mode, is there any way to programmatically get the device's current position relative to the map?

At first, I found that a pose is visible when Visibility...->Debug is enabled, though this appears to be the current odometry pose, and is not localized to the map.

I found discussion question https://github.com/introlab/rtabmap/discussions/970, which appears to almost be what I'm looking for (minus the desktop program part). In RTABMapApp.cpp, I found a call to `SetCameraPose` (https://github.com/introlab/rtabmap/blob/master/app/android/jni/RTABMapApp.cpp#L1432) that is doing something similar, taking a pose combined with            `mapToOdom_`, where `mapToOdom_` is set to `stats->mapCorrection()` elsewhere in the code. Attempting to print out the x, y, and z of `mapToOdom_` after this line, there is essentially no    change in the variable after a successful localization (x = 0.000002, y = 0.000002, y = -0.000004), even though the on-screen map appears to jump to match up the point cloud with the         device's current position.

matlabbe, 2025-04-30:
Check if "Optimize from graph end" is enabled in the Mapping options. If so `mapToOdom_` will stay Identity on re-localization. Disable that option to make odometry pose jump instead of the map moved around the current odometry pose.

sb-msoe, 2025-04-30:
That was the issue, thank you!

closed 2025-04-30 by sb-msoe