introlab/rtabmap #1494: Is MainWindow\geometry ever called for restore?

parkerlreed, 2025-04-28:
I notice the window geometry is saved in the rtabmap.ini

This never seems to properly restore any layouts when opening the program.

Is this supposed to be called at some point or am I missing it somewhere?

```
--
../guilib/src/DatabaseViewer.cpp-       QByteArray bytes;
../guilib/src/DatabaseViewer.cpp-       bytes = settings.value("geometry", QByteArray()).toByteArray();
../guilib/src/DatabaseViewer.cpp-       if(!bytes.isEmpty())
../guilib/src/DatabaseViewer.cpp-       {
../guilib/src/DatabaseViewer.cpp:               this->restoreGeometry(bytes);
--
../guilib/src/CalibrationDialog.cpp-    }
../guilib/src/CalibrationDialog.cpp-    QByteArray bytes = settings.value("geometry", QByteArray()).toByteArray();
../guilib/src/CalibrationDialog.cpp-    if(!bytes.isEmpty())
../guilib/src/CalibrationDialog.cpp-    {
../guilib/src/CalibrationDialog.cpp:            this->restoreGeometry(bytes);
--
../guilib/src/PreferencesDialog.cpp-            settings.beginGroup(window->objectName());
../guilib/src/PreferencesDialog.cpp-            bytes = settings.value("geometry", QByteArray()).toByteArray();
../guilib/src/PreferencesDialog.cpp-            if(!bytes.isEmpty())
../guilib/src/PreferencesDialog.cpp-            {
../guilib/src/PreferencesDialog.cpp:                    window->restoreGeometry(bytes);

```

The other three get called. 

```
[parker@parker-framework build]$ grep geome ~/.rtabmap/rtabmap.ini 
geometry=@ByteArray(\x1\xd9\xd0\xcb\0\x3\0\0\0\0\0\0\0\0\0@\0\0\x4K\0\0\x3{\0\0\0\0\0\0\0@\0\0\x4K\0\0\x3{\0\0\0\0\0\0\0\0\a\x80\0\0\0\0\0\0\0@\0\0\x4K\0\0\x3{)
AboutDialog\geometry=@ByteArray(\x1\xd9\xd0\xcb\0\x3\0\0\0\0\0\0\0\0\0@\0\0\x2\xf7\0\0\x3!\0\0\0\0\0\0\0@\0\0\x2\xf7\0\0\x3!\0\0\0\0\0\0\0\0\a\x80\0\0\0\0\0\0\0@\0\0\x2\xf7\0\0\x3!)
MainWindow\geometry=@ByteArray(\x1\xd9\xd0\xcb\0\x3\0\0\0\0\0\x1\0\0\0H\0\0\x5\0\0\0\x3\x17\0\0\0\x1\0\0\0H\0\0\x5\0\0\0\x3\x17\0\0\0\0\0\0\0\0\a\x80\0\0\0\x1\0\0\0H\0\0\x5\0\0\0\x3\x17)
PreferencesDialog\geometry=@ByteArray(\x1\xd9\xd0\xcb\0\x3\0\0\0\0\0\0\0\0\0\x14\0\0\x3\xf0\0\0\x3\xb0\0\0\0\0\0\0\0\x14\0\0\x3\xf0\0\0\x3\xb0\0\0\0\0\0\0\0\0\a\x80\0\0\0\0\0\0\0\x14\0\0\x3\xf0\0\0\x3\xb0)
```

parkerlreed, 2025-04-29:
Oh I see it defined in PreferencesDialog.cpp, hmm. Trying to trace why this isn't restoring.

parkerlreed, 2025-04-29:
loadMainWindowState appears to be triggered before the main UI even shows. From everything I see this should be after it loads since Qt can overwrite your layout?

[image]

matlabbe, 2025-05-02:
It should restore the geometry and dockwidget states. For example, starting from the default view:

[image]

I add Statistics and Graph View, and resize the window, then ctrl-s to save the changes, then close and re-open rtabmap:

[image]

It is exactly like before I exited. What are the steps you do that don't get saved?

parkerlreed, 2025-05-04:
Okay so adding additional docked widgets does in fact get saved.

What I am unsure of is if the panel sizes get saved?

Resizing the panels triggers the warning "Do you want to save your modified config?" on close, but upon restore they are never the same size.

parkerlreed, 2025-05-04:
https://github.com/user-attachments/assets/fad675ae-835c-4714-bf05-fa0ebf154b8f

parkerlreed, 2025-05-04:
OH, do docks have a default size? The config save DOES affect the sizing but only to a certain extent.

https://github.com/user-attachments/assets/b4f3e5c6-83f4-4123-b226-65c53f488d1d

matlabbe, 2025-05-10:
I reproduced it here. It looks like related to the fullscreen mode. In window mode, the dock widget size is correctly restored. I think it is how the geometry is restored before or after bringing the window in fullscreen.

Based on the doc, we are basically doing the same than this [example](https://doc.qt.io/qt-6/restoring-geometry.html) (in same order):
```
void MyMainWindow::closeEvent(QCloseEvent *event)
{
    QSettings settings("MyCompany", "MyApp");
    settings.setValue("geometry", saveGeometry());
    settings.setValue("windowState", saveState());
    QMainWindow::closeEvent(event);
}
void MainWindow::readSettings()
{
    QSettings settings("MyCompany", "MyApp");
    restoreGeometry(settings.value("myWidget/geometry").toByteArray());
    restoreState(settings.value("myWidget/windowState").toByteArray());
}
```
On close events or when we do "ctrl-s", we save like this:
https://github.com/introlab/rtabmap/blob/6b5b1c96170e6051537d4ab612059dcd44a7a3b5/guilib/src/PreferencesDialog.cpp#L3953-L3958

In the constructor, we restore the window like this:
https://github.com/introlab/rtabmap/blob/6b5b1c96170e6051537d4ab612059dcd44a7a3b5/guilib/src/PreferencesDialog.cpp#L3979-L3988

Related issue: https://forum.qt.io/topic/151253/preserving-restoring-qdockwidget-size

**A possible workaround** is to resize the window to take the fullscreen (without actually clicking fullscreen button), set the dock widget size, then save settings. Close and re-open, the window would appear already in fullscreen mode and dock widget would have the right size.

parkerlreed, 2025-05-11:
Fullscreen here is maximized? I wasn't aware of any actual fullscreen mode. I'll check it out, thanks..

parkerlreed, 2025-05-11:
Looks like boost/cmake updating has killed building this :(

```
-- Found g2o (targets)
-- g2o include dir: /usr/include/g2o/core;/usr/include
-- Latest g2o factory version detected with shared ptr (factory file: /usr/include/g2o/core/factory.h).
CMake Warning (dev) at /usr/share/cmake/Modules/CMakeFindDependencyMacro.cmake:78 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/GTSAM/GTSAMConfig.cmake:21 (find_dependency)
  CMakeLists.txt:525 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
CMake Warning (dev) at /usr/local/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:34 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  CMakeLists.txt:551 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/lib/cmake/Boost-1.88.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /usr/lib/cmake/boost_thread-1.88.0/boost_thread-config.cmake

  but it set boost_thread_FOUND to FALSE so package "boost_thread" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_thread.so.1.88.0 (release runtime, Boost_USE_DEBUG_RUNTIME=TRUE)

  * libboost_thread.a (static, default is shared, set
  Boost_USE_STATIC_LIBS=ON to override)

Call Stack (most recent call first):
  /usr/lib/cmake/Boost-1.88.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake/Modules/FindBoost.cmake:609 (find_package)
  /usr/local/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:34 (find_package)
  CMakeLists.txt:551 (find_package)

-- Configuring incomplete, errors occurred!
```

matlabbe, 2025-05-11:
That error is coming from the included cmake config from libpointmatcher. You may try to rebuild libpointmatcher if you updated boost.
```
Call Stack (most recent call first):
  /usr/lib/cmake/Boost-1.88.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake/Modules/FindBoost.cmake:609 (find_package)
  /usr/local/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:34 (find_package)
  CMakeLists.txt:551 (find_package)
```

parkerlreed, 2025-05-19:
I still haven't been able to find out why boost is broken currently. pointmatcher compiles and installs. Just failing when it comes to building rtabmap.

parkerlreed, 2025-05-19:
@matlabbe It seems like just saving once while in a floating window works! I'm able to maximize, save and reload with the correct positions now.

Thank you so much.