introlab/rtabmap #1495: Help with creating a 3D model

erlendgudm, 2025-04-29:
Hello, i am a student from Norway, and is trying to make a 3D model in RTAB-Map by using pictures which i have gotten from the company that i am doing the task for. They have sent me photos taken by a stereo camera with a baseline of 0.1701m. The camera type is Voyis. I have seperated the pictures in to left and right, and tried to calibrate it. It works for a few pictures, and then the backround colour becomes red. Is there anyone who maybe could help me out, or maybe tell me what the problem can be? It seems to be losing the odometry connection after about 10 pictures, but i can se clear points in the 3D MAP ribbon, but when the picture "session expires", the cloud doesnt change, and has almost no points.

Thank you in advance.

erlendgudm, 2025-04-29:
[image]

This is how the screen looks after some pictures

muhammedali99, 2025-04-30:
one thing you may try is removing the pic no'10 or 11 that may be causing this probleme , if there is a big diffrence between a image and the one next (too much movement) to it rtab-map will fail to make motion estimation which will fail the odometry 

some things to check:

Reg/Strategy = 1 
RGBD/OdomSensorSync =  true
Odom/Strategy  = 0

but using a product like Voyis with a build in IMU sensor , make sensor fusion with CSV data from imu and vd/images to estimate the position 

try using rtabmap-ros and make a ros2 bag with ur data and then run it with 
rtabmap-ros , it should be much easier

matlabbe, 2025-05-02:
@erlendgudm Can you share a database? even if there are only a couple of images in it? 

> tried to calibrate it.

How did you try to calibrate it? For stereo to work well, a good stereo calibration is required and the left/right images should be exactly synchronized. I would expect for commercial stereo cameras that they are already calibrated.