introlab/rtabmap #1500: How to link against pre-compiled binaries (include/lib) from Releases on Windows?

DoraemonMiku, 2025-05-06:
Hi RTAB-Map developers,
I'm new to RTAB-Map and C++ development on Windows. Compiling from source seems difficult for me right now.
I downloaded the pre-compiled release from the GitHub Releases page. It has include, lib, and bin folders.
How can I configure my simple CMake project to use these pre-compiled files?

DoraemonMiku, 2025-05-06:
I tried to install and compile RTABMap using vcpkg, but the resulting package has incomplete functionality; CUDA support was not enabled, and neither was K4A driver support.
I want to get real-time odometry information in other Windows programs.
I was wondering if you could shed some light on how the release version RTABMap-0.21.4-win64-cuda117.7z (from [https://github.com/introlab/rtabmap/releases/download/0.21.4/RTABMap-0.21.4-win64-cuda117.7z](https://www.google.com/url?sa=E&q=https%3A%2F%2Fgithub.com%2Fintrolab%2Frtabmap%2Freleases%2Fdownload%2F0.21.4%2FRTABMap-0.21.4-win64-cuda117.7z)) was compiled, especially regarding the enabling of CUDA and K4A drivers?
Any help or pointers would be greatly appreciated!

matlabbe, 2025-05-11:
We would probably need to remove the include and lib folders from the binaries, as they cannot really be used alone. The VCPKG approach should be used if you need to use rtabmap as a library. To build the new release, I use vcpkg to install all dependencies, then I build rtabmap outside vcpkg with the options I need, or with more dependencies (like CUDA or k4a). Based on https://github.com/introlab/rtabmap/wiki/Installation#source-1

```bash
# added opencv with cuda here
vcpkg install ceres g2o octomap zlib sqlite3 opencv[contrib,nonfree,cuda] pcl[qt,vtk] yaml-cpp --clean-after-build

git clone https://github.com/introlab/rtabmap.git
cd rtabmap/build
cmake -DCMAKE_TOOLCHAIN_FILE=C:/vcpkg/scripts/buildsystems/vcpkg.cmake -DWITH_K4A=ON ..
cmake --build . --config Release
# To install library and use it in another project. There would be a `RTABMapConfig.cmake` generated.
cmake --build . --config Release --target install
```
For K4A, just install the [K4A binaries](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#installation), cmake should be able to find it when doing the cmake command above.

To use in another cmake project:
```bash
cmake -DRTABMap_DIR="C:/Directory/where/RTABMapConfig.cmake/is/installed" ..
```

Note: to avoid g2o linking errors with last vcpkg version, make sure to use that patch: https://github.com/microsoft/vcpkg/pull/45000

DoraemonMiku, 2025-05-12:
> We would probably need to remove the include and lib folders from the binaries, as they cannot really be used alone. The VCPKG approach should be used if you need to use rtabmap as a library. To build a new release, I use vcpkg to install all dependencies, then I build rtabmap outside vcpkg with the options I need, or with more dependencies (like CUDA or k4a). Based on https://github.com/introlab/rtabmap/wiki/Installation#source-1
> 
> # added opencv with cuda here
> vcpkg install ceres g2o octomap zlib sqlite3 opencv[contrib,nonfree,cuda] pcl[qt,vtk] yaml-cpp --clean-after-build
> 
> git clone https://github.com/introlab/rtabmap.git
> cd rtabmap/build
> cmake -DCMAKE_TOOLCHAIN_FILE=C:/vcpkg/scripts/buildsystems/vcpkg.cmake -DWITH_K4A=ON ..
> cmake --build . --config Release
> # To install library and use it in another project. There would be a `RTABMapConfig.cmake` generated.
> cmake --build . --config Release --target install
> For K4A, just install the [K4A binaries](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#installation), cmake should be able to find it when doing the cmake command above.
> 
> To use in another cmake project:
> 
> cmake -DRTABMap_DIR="C:/Directory/where/RTABMapConfig.cmake/is/installed" ..
> Note: to avoid g2o linking errors with last vcpkg version, make sure to use that patch: [microsoft/vcpkg#45000](https://github.com/microsoft/vcpkg/pull/45000)

Thank you so much for your clear explanation and the build instructions! I followed them, and it successfully resolved my problem. Much appreciated!

closed 2025-05-12 by DoraemonMiku