introlab/rtabmap #1510: no transform between frames map and base_link

m11212044, 2025-05-22:
Thank you for open-sourcing this project.

I am a student learning to use RTAB-Map.

Following #1221 and #531, I successfully set up the environment required for SuperPoint in RTAB-Map and tested my RTAB-Map using the following commands:

```
roslaunch realsense2_camera rs_camera.launch align_depth:=true

roslaunch rtabmap_launch rtabmap.launch \
    rtabmap_args:="--delete_db_on_start 
    --SuperPoint/ModelPath /home/ros/Launch/superpoint.pt
    --Kp/DetectorStrategy 11" \
    database_path:=~/visual_mapflie/superpoint_slam_test3.db \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false
```

The test result was successful, and the map was correctly generated.

However, when I ported it to my own robot and launched it using the attached launch file, I encountered the following error message:

`[warn] [1747905219.480990764]: no transform between frames map and base_link available after 20.004472 seconds of waiting. this warning only prints once.`

May I ask what could be the reason for this?

[mylaunch.txt](https://github.com/user-attachments/files/20390759/mylaunch.txt)

P.S. If I set both Kp/DetectorStrategy and Vis/FeatureType to "4", the mapping works fine again.

yuwei200089, 2025-05-26:
I have resolved the issue by regenerating the `Superpoint.pt` file.

closed 2025-05-26 by yuwei200089

matlabbe, 2025-05-27:
It felt that there was an error happening on the robot. Glad that you found it. For others that wondering how regenerating `superpoint.pt`, we can use this [script](https://github.com/introlab/rtabmap/blob/master/archive/2022-IlluminationInvariant/scripts/trace.py) (run it on same computer that will be running rtabmap to make sure the version is compatible).