introlab/rtabmap #1531: Is it possible to perform RGB-D SLAM using pre-recorded RGB-D video without IMU data?

NUHashikou, 2025-06-11:
Thank you for open-sourcing this excellent repository. I am currently using the precompiled binary of RTAB-Map 0.22.0 for Windows (win64) that you have kindly provided.

Now, to the main point — I would like to ask whether it is possible to perform SLAM on pre-recorded RGB-D data captured at 30 fps, similar to how it is done in the iOS app. Unfortunately, IMU data is not available.

I believe my case is similar to [issue #1501](https://github.com/introlab/rtabmap/issues/1501). I have modified several settings in Preferences, and recorded using RGB and depth images along with camera intrinsic parameters. However, odometry is not computed beyond the second frame.

If it is possible to perform SLAM in such a setup, I would greatly appreciate your guidance. I have attached the .ini file that I am currently using.

[RTAB-Map_config_for_RGBD-SLAM.txt](https://github.com/user-attachments/files/20689807/RTAB-Map_config_for_RGBD-SLAM.txt)

matlabbe, 2025-06-13:
Can you share a subset of your dataset?

But yes, it is possible to use a directory of RGB images, a directory of corresponding depth images, with camera calibration file (yaml). You can select `RGB-D` source type, then `Images` driver. Back in 2015, we added an example with [RGB-D Tum dataset](https://cvg.cit.tum.de/data/datasets/rgbd-dataset/download): http://official-rtab-map-forum.206.s1.nabble.com/How-to-process-RGBD-SLAM-datasets-with-RTAB-Map-tp939p647.html

NUHashikou, 2025-06-16:
Dear matlabbe,

Thank you very much for taking the time to respond, even though I asked a redundant question about something already known. My dataset is essentially the same as TUM's, with only slight formatting differences. Following the link you kindly provided, everything worked perfectly!

The issue was that I had described the camera's internal parameters incorrectly. I sincerely appreciate your support.

closed 2025-06-16 by NUHashikou