introlab/rtabmap #1537: Error processing multiband via CLI

antoinemacia, 2025-06-16:
Hello!

I seem to have an issue when running Multiband via the export CLI, a few more info

- I use the `introlab3it/rtabmap:focal` Docker image which has AliceVision built in
- My database reprocessing is as bellow, coming from iPhone Pro LIDAR via app set to data capture mode (please feel free to suggest different parameters if those make not sense for the capture device, I believe I took those from a forum post)

```
rtabmap-reprocess -default \
    --Mem/UseOdomFeatures false \
    --Kp/DetectorStrategy 1 \
    --Vis/FeatureType 1 \
    --RGBD/OptimizeMaxError 8 \
    --Vis/MinInliers 15 \
    --Mem/DepthAsMask false \
    --Vis/CorNNDR 0.6 \
    --Kp/NndrRatio 0.6 \
    "$DB_FILE_ABS" "${PREFIX}_processed.db"
```

- My export parameters
```
rtabmap-export \
    --texture \
    --mesh \
    --multiband \
    --uwarn \
    --multiband_fillholes \
    --texture_size 4096 \
    --texture_range 3 \
    --poisson_depth 9 \
    --max_polygons 0 \
    --color_radius 0 \
    --no_clean \
    --output_dir . \
    "$PROCESSED_DB"
```

- My log
```
Texturing... done (13.428884s).
Merging 698 texture(s) to single one (multiband enabled)...
Merging to 1 texture(s)... done (7.187190s).
MultiBand texturing (size=4096, downscale=2, unwrap method=Basic, fill holes=true, padding=5, best score thr=0.100000, angle thr=90.000000, force visible=true)... "./250605154648_data__processed_mesh_multiband.obj"
Failed to find match for field 'rgb'.
/app/src/workflows/plann3d/build-simple-rtab.sh: line 107:  5920 Illegal instruction   rtabmap-export --texture --mesh --multiband --uwarn --multiband_fillholes --texture_size 4096 --texture_range 3 --poisson_depth 9 --max_polygons 0 --color_radius 0 --no_clean --output_dir . "$PROCESSED_DB"
```

Is thisLet me know if you need the database for debugging purposes as well and I can put it into a drive

Thank you!

matlabbe, 2025-06-16:
On which host computer architecture are you running it? "Illegal instruction" sounds like amd64 running on arm64 CPU or the inverse.

Just tried it on some database, no issue.
```
docker run -it --rm \
   -v /home/mathieu/Documents/RTAB-Map:/tmp/host \
   introlab3it/rtabmap:focal \
   rtabmap-export \
      --texture \
      --mesh \
      --multiband \
      --uwarn \
      --multiban_fillholes \
      --texture_size 4096 \
      --texture_range 3 \
      --poisson_depth 9 \
      --max_polygons 0 \
      --color_radius 0 \
      --no_clean \
      /tmp/host/241006-162609.db
```

antoinemacia, 2025-06-16:
AH ofc! Running this on ARM, good to close thank you for looking into it! Ill investigate if using multiband_fillholes is worth it

matlabbe, 2025-06-17:
Still surprised that it crashed just when reaching multiband code and not before. For the fillholes option, I created a follow-up issue: https://github.com/introlab/rtabmap/issues/1541