introlab/rtabmap #1550: Mesh -> Gaussian splats

jitesh3023, 2025-07-02:
Is there a way to use the reconstructed mesh to create Gaussian splatting? I believe that using the images captured during mesh scanning may not be ideal for training Gaussians. During mesh scanning, we tend to scan meticulously by moving closer to objects, whereas for Gaussian splatting, it is better to scan from a distance to capture more information in a single view. This wider coverage helps improve overlap detection when running COLMAP. Is there any way to generate Gaussian splatting from the data collected during mesh scanning? I tried using the camera poses and RGB images from db file for training gaussians, but the results were bad. I would appreciate it if anyone knows a method to achieve this or is aware of any open-source repository that supports this workflow.

jitesh3023, 2025-07-07:
Thank you so much @elidiot18 for the detailed workflow and clear explanation — I will definitely give it a try.

I just had a quick question: have you tried using RGB images captured during mesh scanning with RTAB-Map for Gaussian Splatting? I'm unsure if they provide enough viewpoint diversity and parallax for high-quality GS results. Since mesh scanning typically involves slower, close-range motion, I’m concerned it might impact the reconstruction quality. If you have tried this, I’d love to know how the results turned out.

elidiot18, 2025-07-07:
Yes I have tried, and I would say that it depends how thorough you were while scanning in the first place. If you scan quickly in order to get an ok mesh, the images will be too blury and probably not good enough for a perfect result. But it depends on the camera you're using. Also you cannot expect to magically reconstruct all details if you didn't take pictures of them from different angles.
Depends strongly on your scene and how you scanned.

The results I had were pretty nice but also not "perfect", but that can be due to the fact that I'm working on dark environments. I'm still experimenting with it right now.