introlab/rtabmap #1556: Consultation  on iOS-to-Desktop Post‑Processing Workflow in RTAB‑Map

samuelyhsu, 2025-07-12:
Hello,

First of all, thank you very much for open‑sourcing the RTAB‑Map project.

I’m looking to extend this work to detect static objects of interest and obtain their coordinates. Here’s the workflow I have in mind:

- Perform data capture on the iOS app and export the database file;
- Post‑process that database on the desktop to export camera poses and otherdata;
- Use the camera poses and other data to compute the coordinates of my target objects.

I have a few questions:

- **iOS capture parameters:** Which settings can speed up data capture (real‑time mapping isn’t required) without significantly degrading post‑processing pose accuracy?
- **Desktop post‑processing parameters:** To achieve the highest possible global pose accuracy, which parameters do you recommend to focus?
- **Batch export of Node & Data:** I would like to export `{ Node {id, pose}, Data {id, image, depth, depth_confidence, calibration} }` in one go—does a built‑in command or configuration already support this, or must I extend the code myself?

If you have the time to share any experience or suggestions, ,it would be greatly appreciated.

Thank you again.

matlabbe, 2025-07-15:
For the VIO accuracy, we cannot change it, it is already working as fast as it can. To record more data in the database, you can increase the frame rate, though if the map increases fast in size, it may take more and more time to add new frames, frames could be skipped. We added a Data Recorder mode, which will directly record all frames at the specified frame rate without overhead. However, that mode won't run loop closure detection online so you won't have any feedback if you correctly close the loops. When recording data with Data Recorder mode, to do post-processing on Desktop afterwards, see the command used [here](http://official-rtab-map-forum.206.s1.nabble.com/iOS-best-practices-for-only-tracking-long-recordings-tp10300p10998.html) or [here](https://github.com/introlab/rtabmap/issues/1320#issuecomment-2403737157).

For the VIO part, we cannot improve it in post-processing. The only that could be improved is loop closure detection. Note that if loop closures cannot be detected (there is actually no overlap in the recorded data), then nothing can be done to improve the accuracy of the trajectory. Other thing that can be done in post-processing to improve the map is sometime ArKit will do a very bad drift, then in `rtabmap-databaseViewer` it is possible to remove or refine those constraints (open Graph View and add/remove constraints with Constraints View). 

For the exported data, these can be all exported with Export Poses or Export Images menu options on rtabmap-databaseViewer. It could be also done with `rtabmap-export` tool: 
```
# The calibration file will include the transform from base to camera optical frame
rtabmap-export --poses --poses_format 11 --images_id rtabmap.db

# Export poses directly in camera optical frame
rtabmap-export --poses_camera --poses_format 11 --images_id rtabmap.db
```

cheers,
Mathieu

samuelyhsu, 2025-07-19:
Thank you so much, I will try it.

closed 2025-07-19 by samuelyhsu