introlab/rtabmap #1571: CameraRealSense2::init() fails for realsense D405

gkoundry, 2025-08-27:
I'm using the C++ API (no ROS) and when I call `camera = new CameraRealSense2("D405");` it fails with the following error:
```
[ INFO] (2025-08-27 09:07:23.280) CameraRealSense2.cpp:617::init() Device Name: Intel RealSense D405
[ INFO] (2025-08-27 09:07:23.280) CameraRealSense2.cpp:620::init() Device FW version: 5.16.0.1
[ INFO] (2025-08-27 09:07:23.280) CameraRealSense2.cpp:623::init() Device Product ID: 0x0B5B
[ INFO] (2025-08-27 09:07:23.280) CameraRealSense2.cpp:633::init() Device Sensors: 
[ INFO] (2025-08-27 09:07:23.280) CameraRealSense2.cpp:701::init() Stereo Module was found.
terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "sensor"
Aborted (core dumped)
``` 

It is failing because the code expects both a RGB camera and stereo module but the D405 does not have a RGB sensor (see https://github.com/IntelRealSense/librealsense/issues/11981).

matlabbe, 2025-09-03:
It seems crashing in that loop:
https://github.com/introlab/rtabmap/blob/52e770e2d46a95c739c1890bd499f2a9f2912e61/corelib/src/camera/CameraRealSense2.cpp#L625-L694

If you can change the source code, can you move the `UINFO("%s was found.", elem.get_info(RS2_CAMERA_INFO_NAME)); ` at the beggining of the loop to see in which module it is crashing?

A workaround is to launch the D405 in IR mode, either IR-Depth or Stereo-IR by using these functions:
https://github.com/introlab/rtabmap/blob/52e770e2d46a95c739c1890bd499f2a9f2912e61/corelib/include/rtabmap/core/camera/CameraRealSense2.h#L75-L76

```cpp
camera->setEmitterEnabled(false);
camera->setIRFormat(true, true);
```

gkoundry, 2025-09-09:
For the D405 you have one sensor with steam profiles for both RGB and depth.  Rtabmap does not seem to handle this case. I was able to make some changes to support getting both  depth and RGB from the same sensor:
```
--- a/corelib/src/camera/CameraRealSense2.cpp
+++ b/corelib/src/camera/CameraRealSense2.cpp
@@ -699,7 +699,7 @@ bool CameraRealSense2::init(const std::string & calibrationFolder, const std::st
        std::vector<std::vector<rs2::stream_profile> > profilesPerSensor(sensors.size());
         for (unsigned int i=0; i<sensors.size(); ++i)
         {
-               if(i==0 && ir_ && !stereo)
+               if(i==0) // && ir_ && !stereo)^M
                {
                        continue;
                }
@@ -760,7 +760,7 @@ bool CameraRealSense2::init(const std::string & calibrationFolder, const std::st
                                                                intrinsic.model,
                                                                intrinsic.coeffs[0], intrinsic.coeffs[1], intrinsic.coeffs[2], intrinsic.coeffs[3], intrinsic.coeffs[4]);
                                                added = true;
-                                               if(video_profile.format() == RS2_FORMAT_RGB8 || profilesPerSensor[i].size()==2)
+                                               if(profilesPerSensor[i].size()==2)^M
                                                {
                                                        break;
                                                }
```

Obviously this only is valid for the D405 but hopefully it helps.