introlab/rtabmap #1574: Empty /mapData, but /map generates correctly?

16stelter, 2025-09-03:
Heya, 

I'm new to rtabmap and am trying to set it up to run slam in a 2.5d hill environment with few walls. I have a tof camera aimed at the ground in front of the robot at a 45 degree angle. I start the rtabmap_slam/rtabmap node and give that pointcloud to rtabmap via /scan_cloud. On the /map topic I can see the 2d occupancy map being created correctly, but I would also like to see (and further process) the 3d map. I've seen in other demos that they were able to visualize the 3d map from the /mapData topic. However, if I visualize that topic in rviz2, it shows messages being received, but points 0 and nodes 0 shown of 0. I'm probably missing some configuration settings, but I'm not sure what I need to change. Currently, I set the following parameters:

```
rtabmap:
  ros__parameters:
    frame_id: "base_link"
    subscribe_depth: false
    subscribe_rgb: false
    subscribe_scan: false
    subscribe_scan_cloud: true
    approx_sync: true
    sync_queue_size: 50
    topic_queue_size: 30
    qos: 2
    #use_action_for_goal: true
    Reg/Strategy: "1"
    Reg/Force3DoF: "false"
    RGBD/NeighborLinkRefining: "true"
    Grid/RangeMin: "0.2"
    Grid/RangeMax: "3.0"
    Grid/3D: "true"
    Grid/FromScan: "true"
    Optimizer/GravitySigma: "0"
```

Everything else should be the default values. Thanks for any help!

matlabbe, 2025-09-06:
You would have to check Cloud from scan" checkbox under MapCloud rviz plugins.

[image]

16stelter, 2025-09-08:
Ah okay yes that fixes that problem, thank you very much! I now have the problem that the pointcloud seems to drift downwards over time (see screenshot), even though the IMU readings from my robot show mostly flat. Is there anything I need to set to correct that?

[image]

matlabbe, 2025-09-10:
Maybe the static transform between the IMU and the tof camera is wrong. Can you share the database?

matlabbe, 2025-09-21:
To share bigger databases, people often share a public link to a file in a google drive or dropbox.

How is the odometry computed? The curvy path is caused by ICP registration (`RGBD/NeighborLinkRefining: "true"`). Here is an example of the consecutive frames using input odometry:

[image]

Here the same frames but after `RGBD/NeighborLinkRefining` is done:
[image]

The point cloud now overlaps nicely, but it adds significant roll. The error may come from the depth sensor, why the surface is so curvy (while it should not?)? Note that even if you fix the depth distortion, ICP refining is not very usable on a plane like that (to work well, ICP needs at least point normals in the 3 main planes, not only one). I would suggest to turn `RGBD/NeighborLinkRefining` off.

matlabbe, 2025-09-28:
That [paper](https://arxiv.org/abs/2403.06341) is kinda the main reference. In particular for the issue above, see section "3.1.2 Lidar Odometry" and the paragraph starting with "Motion Prediction:".

closed 2025-09-28 by matlabbe