introlab/rtabmap #1584: Global loop closure when set _maxMemoryAllowed

azulejosss, 2025-09-18:
Hello, may I ask if the global loop closure cannot detect the loop signature that have been removed from the wm region when I set _maxMemoryAllowed? 
For instance, if I have the robot walk a long way from the starting point and then return to it, since I have limited the number of signatures in the wm region, the first signature of the loop frame has already been removed from the wm region. Therefore, the global loop detection of rtabmap cannot detect the loop of this frame. Is this understanding correct.

matlabbe, 2025-09-21:
Yes, loop closure detection happens only with remaining nodes in WM.

A trick to keep the first node in the map a longer time is to wait there a little (staying still a couple of seconds) to build up the weight of the node, then it would not be transferred first to LTM when the max memory allowed is reached. This [paper](https://arxiv.org/abs/2407.15305) discusses similar events that could happen, for which we should aware of when enabling memory management.

azulejosss, 2025-09-22:
Thank you for your explanation. 
I noticed that before performing the global loop detection, there is a step of radius retrieval from the LTM. However, in the code, the retrieval is carried out from the pose map after memory management. If I change this to retrieval from all nodes, then when the robot returns to the vicinity of the starting point, it will add the starting point back to the WM, thereby achieving long-distance loop detection. May I ask if this method is feasible.

matlabbe, 2025-09-28:
> If I change this to retrieval from all nodes, then when the robot returns to the vicinity of the starting point, it will add the starting point back to the WM

Once a node is transferred to LTM, it is not linked anymore to current local graph in WM (nodes in LTM would drift over time from the nodes in WM). We therefore we cannot "retrieve from LTM nodes close to current location". To be able to retrieve based on position from LTM, we would need to keep the global graph in WM. In localization mode, that could be a nice option to have because we don't need to re-optimize the global graph anymore after initialization and the graph itself is not using a lot of RAM: see also https://github.com/introlab/rtabmap/issues/1245. For SLAM mode, it is trickier because global graph optimization's processing time increases with the size of the map, which is against the "constant" memory / cpu usage of the memory management approach.